sinara-hw/sinara #582: remove Si5324 to HMC830 connection/mux on the RTM

sbourdeauducq, 2018-08-17:
https://github.com/sinara-hw/sinara/issues/576#issuecomment-412284694

gkasprow, 2018-11-27:
removed, together with mux.
@hartytp what we do with LO signal routing to all mezzanines?

hartytp, 2018-11-28:
@gkasprow On Sayma v2.0 there will not be a LO generated on the RTM because there will not be a clock mezzanine.

The plan is to route the reference to each AFE. The AFEs can then have a PLL if required.

gkasprow, 2018-11-28:
you mean 100MHz reference we get from SMA input?

hartytp, 2018-11-28:
@gkasprow it will usually be 125MHz in v2.0 to make the synchronisation easier, but yes, that's what I mean.

gkasprow, 2018-12-30:
We will use ADCLK948 for reception of SMA ref signal and distribution of it to mezzanines (LO signal) so we can leave Si5324 connection to ADCLK948 with no cost.

marmeladapk, 2019-01-11:
I guess this issue can be closed. Added benefit of keeping this connection is that we can use WR clock if we decide to implement it on RTM.

closed 2019-01-11 by marmeladapk