sinara-hw/sinara #583: DAC synchronisation in Sayma v2.0

hartytp, 2018-08-30:
Proposal for DAC clocking and synchronization in Sayma v2.0:
- users supply the RTM with a reference clock (frequency `f_ref`) via the front panel (FP) SMA this will be the only supported clocking path (until/unless we adopt some fancy WR-type PLL, which can distribute a high-quality clock to the RTM via DRTIO)
- users supply the AMC with a RTIO clock (frequency `f_rtio`) either via FP SMA (master/standalone) or via fibre + CDR (satellite)
- in the first instance, the only configuration supported in gateware/firmware (NB this is not a hw limitation) will be `f_ref = f_rtio` to minimize complexity. In future it won't be too hard to support different configurations, such as `f_ref=100MHz` `f_rtio=125MHz`. NB the cost of a really low-noise PLL from Wenzel or similar to generate this clock is small compared with the overall cost of ownership of Sayma
- the reference and rtio clocks must be phase locked together (e.g. by locking to a common reference)
- the RTM reference clock will be buffered with a low-noise buffer (e.g. ADCLK925) and sent to the AMC via the AMC<->RTM connector, as well as an HMC830. This will be useful for future diagnostics. (Unless you don't think this is worth it @sbourdeauducq)
- All clocks and important signals on the AMC + RTM will be accessible via suitable test points, such as UFL connectors!
- a HMC830 on the RTM generates the DAC clock at up to 2.4GHz by multiplying the reference clock (NB for integer-N synthesis, the input-output propagation delay is deterministic)
- the HMC830 output is buffered (again, ADCLK925 or similar) and fed directly to the DACs. There is a constant propagation delay/phase offset (limited only by drift in PLLs and buffers) between the DAC clocks and the reference/rtio clocks. The is no frequency division here, so there is no non-deterministic phase offset
- To synchronize the DACs to the RTIO clock, all that needs to be done is to supply a SYSREF pulse from the AMC FPGA that meets S/H at the DACs (w.r.t. the DAC clock, which can run at up to 2.4 GHz)
- In principle, the SYSREF pulse could be generated directly by the FPGA. However, the S/H window for a 2.4GHz DAC clock is something like 200ps, which will be hard to achieve for a variety of reasons: drifts between the phase of the RTIO and reference clock due to the SI5324 (already something like 100ps); FPGA jitter; SI between the two boards; general VT fluctuations in all components; etc
- As a result, we opt to put a discrete, low jitter FF on the RTM to retime the SYNC pulse produced by the FPGA. This FF is clocked by the RTM reference clock. This ensures that the FPGA is able to generate a low-jitter SYNC pulse with a fixed phase relationship to the RTM reference clock
- since the input timing window for this is close to 8ns, meeting S/H timing is relatively easy (we could nonetheless choose to buffer these signals and route a copy to the RTM FPGA for diagnostics if @sbourdeauducq thinks it's worth it?)
- in principle, we could fix the time delay between the FF output and the DAC SYSREF pin by trace length to ensure that S/H timings are always met in HW. However, given how narrow the timing window is, I'd be inclined to add a phase trim to this. The easiest way of achieving this is with a very simple digital delay line placed after the FF, such as http://www.onsemi.com/PowerSolutions/parametrics.do?id=586

Comments?

hartytp, 2018-08-30:
Other notes on this:
1. We will no longer have a clock for the JESD204B ADCs. This is fixable (e.g. add a second HMC830) but, given that there don't actually seem to be any users of the JESD ADCs, I'd prefer to get rid of them. (If it's a choice between simple, reliable DAC synchronisation and JESD ADCs, I take the JESD ADCs ever time). In any case, users can always add ADCs to the AMC FMC port
2. Need to buy a phase shifter eval board and measure the propagation delay temp co to check that there won't be issues there (highly unlikely, but needs testing)
3. NB the idea of the above is that we keep the bits of phase synchronisation that already work well in ARTIQ (programming the HMC830 and setting the analog phase of the SYSREF to meet S/H at the DAC), but replace the HMC704x with a simpler solution that isn't bogged down with buggy features that we don't want

If everyone (@sbourdeauducq @gkasprow @jbqubit and anyone else who still cares) is happy with this proposal, we can discuss whether it's possible to retrofit this synchronisation setup onto Sayma v1.0 for testing

hartytp, 2018-08-30:
Let's go for NB6L295M as the phase shifter unless anyone has a better idea. NB this has two channels, so we can trim the DAC sysref phases independently. Not sure if that's actually necessary, but might as well give ourselves the option

hartytp, 2018-08-30:
@gkasprow any preference/suggestion for which FF to use?

hartytp, 2018-08-30:
Unless anyone else has a better suggestion, let's use MC100EP51 (I really like ON for this kind of discrete logic). The SOIC version should be fine for dead-bugging/prototyping.

dhslichter, 2018-08-30:
> All clocks and important signals on the AMC + RTM will be accessible via suitable test points, such as UFL connectors!

:+1: 

> We will no longer have a clock for the JESD204B ADCs. This is fixable (e.g. add a second HMC830) but, given that there don't actually seem to be any users of the JESD ADCs, I'd prefer to get rid of them. (If it's a choice between simple, reliable DAC synchronisation and JESD ADCs, I take the synchronization scheme ever time). In any case, users can always add fast ADCs to the AMC FMC port

:+1: 
I think getting rid of the JESD204B ADCs is the right thing to do.  @gkasprow has preferences in terms of ripping out traces vs. simply DNP, and I defer to his preference on that subject.  

> Let's go for NB6L295M as the phase shifter 
> let's use MC100EP51 

Both of these parts look suitable to me.

One other question - how do we verify we are meeting S/H on the DACs for the SYSREF, i.e. is there a clean mechanism to calibrate the phase shifter/delay line that can be done without the need for accessing test points on the board?  

> Edit: we could consider adding solder jumpers to allow the FF + phase shifter to be bypassed in Sayma v2.0 if anyone wants to have a go at driving SYSREF directly from the FPGA.

Given the simplicity of the FF + phase shifter solution, and the known problems with trying to keep drifts from ruining your timing window if driving SYSREF directly over the RTM connector from the FPGA, I can't imagine a use case where one would prefer to drive directly from the FPGA rather than through FF + phase shifter.  I think the phase shifter is definitely needed/desired, it's simple SPI control, we validate that it works sensibly in a dead bug setup...

hartytp, 2018-08-30:
> I think getting rid of the JESD204B ADCs is the right thing to do. @gkasprow has preferences in terms of ripping out traces vs. simply DNP, and I defer to his preference on that subject.

Well, note that if we follow this proposal there will no longer be an ADC clock/SYSREF generation. So, getting the ADCs to function requires more than just populating them, the need a new clock synth/divider/etc.

> One other question - how do we verify we are meeting S/H on the DACs for the SYSREF, i.e. is there a clean mechanism to calibrate the phase shifter/delay line that can be done without the need for accessing test points on the board?

Yes, the DAC has sysref validation built in. We currently use this in ARTIQ and it works fine (all that needs to be done is a fw switch to move from using the analog phase shifters in the HMC7043 to the SPI delay line, but that's trivial).

hartytp, 2018-08-30:
Here is a pretty shoddy illustration of what I have in mind, in case it helps...

[image]

hartytp, 2018-08-30:
Plans for testing this on the hw:
- I've ordered a phase shifter eval board as well as a couple of those FFs in the SOIC package
- I'll dead bug a FF onto the phase shifter eval board
- i'll supply the clock for the FF from my main 150MHz clock synth using a power splitter
- I'll program the HMC7043 as a normal buffer (fundamental mode + no analog phase shift) so it's not taking any part in the synchronisation and has completely deterministic latency
- I'll disconnect the HMC7043 SYSREF signals by DNPing the AC coupling caps
- I'll solder coax pig tails onto the cap pads and use that to hook the phase shifter to the DAC sysref inputs
- I'll connect the phase shifter SPI bus to the RTM somewhere. Probably using the clock mezzanine header (unless @gkasprow can suggest some more easily accessible pins on the AMC/RTM)
- I'll do a really quick check of the phase stability of the phase shifter using a hot air gun and scope (remember SYSREF only needs to be stable to <100ps or so, so the stability isn't crucial)
- I'll see if I can get the SYSREF working with this setup

How does that all sound?

sbourdeauducq, 2018-08-30:
> the RTM reference clock will be buffered with a low-noise buffer (e.g. ADCLK925) and sent to the AMC via the AMC<->RTM connector, as well as an HMC830. This will be useful for future diagnostics. (Unless you don't think this is worth it @sbourdeauducq)

The FPGA needs a clock for the JESD transceivers and there are some constraints as you noticed. We might need a 2X (250MHz) clock.

> since the input timing window for this is close to 8ns, meeting S/H timing is relatively easy

Someone has to check datasheets / experiment / validate this but we can perhaps make the FPGA scan a delay on a short pulse (less than one period of the flip-flop clock) while checking if the DAC receives any SYSREF. Then the timing window at the FF can be derived.

> As a result, we opt to put a discrete, low jitter FF on the RTM to retime the SYNC pulse produced by the FPGA. This FF is clocked by the RTM reference clock. This ensures that the FPGA is able to generate a low-jitter SYNC pulse with a fixed phase relationship to the RTM reference clock

The FF could also absorb the timing variation due to place-and-route, which is very high even when using the global clock network with those annoying Ultrascale chips.

hartytp, 2018-08-30:
> The FPGA needs a clock for the JESD transceivers and there are some constraints as you noticed. We might need a 2X (250MHz) clock.

Yes, I was in the middle of writing exactly that when you posted. Can we use use the Si5324 to provide the transciever ref clock?

hartytp, 2018-08-30:
> Someone has to check datasheets / experiment / validate this but we can perhaps make the FPGA scan a delay on a short pulse (less than one period of the flip-flop clock) while checking if the DAC receives any SYSREF. Then the timing window at the FF can be derived.

What exactly are you worried about? That FF has well-specified timing windows, with S/H of 100ps. For a 125MHz clock, that's a pretty huge window to hit, even when SI etc are taken into account (whatever its failings might be, the SI on Sayma has proved to be excellent).

> The FF could also absorb the timing variation due to place-and-route, which is very high even when using the global clock network with those annoying Ultrascale chips.

True, that's nice. So long as those variations are small compared to the 8ns timing window of the FF these can be absorbed.

sbourdeauducq, 2018-08-30:
> For a 125MHz clock, that's a pretty huge window to hit,

Things always go wrong on Sayma! Without precautions, by default it will hit that 100ps window, guaranteed!

> Can we use use the Si5324 to provide the transciever ref clock?

No, the Si5324 provides the DRTIO transceiver clock already.

But that 2X clock, if we actually need it (someone has to dig into the Xilinx docs), does not need any phase synchronization and can be produced with a basic divider on the 2.4Ghz clock.

hartytp, 2018-08-30:
Side note: this scheme is conceptually a bit ugly since we're rebuilding about 30% of the HMC7044's functionality for about 300% of the cost (but, still nothing compared with the total cost of Sayma) using discrete logic. But, it has the big advantage that we now have simple building blocks that can be probed and debugged independently rather than a poorly documented black box with known bugs. Ugly, but practical.

hartytp, 2018-08-30:
> >No, the Si5324 provides the DRTIO transceiver clock already.
> Does it not have enough outputs to do both?

If not, is it worth using one of the chips in that family that has more outputs? Might be simpler and cheaper than using a separate divider. Anyway, let's do this whichever way you prefer.

dhslichter, 2018-08-30:
> But that 2X clock, if we actually need it (someone has to dig into the Xilinx docs), does not need any phase synchronization and can be produced with a basic divider on the 2.4Ghz clock.

Or a 2x doubler of the RTIO clock, if that's simpler...

hartytp, 2018-08-30:
> Or a 2x doubler of the RTIO clock, if that's simpler...

Lots of ways of skinning a cat. That would also work...

Since the GTH PLLs are not very flexible and have a few non-obvious constraints, it's nice to have a programmable reference frequency. So, we probably want a switchable divider/multiplier here.

FWIW, my feeling is that the easiest way of doing this would probably be to replaced the Si5324 with a model with more outputs, such as a Si5366, which is essentially the same cost as the Si5324. Anyway, let's leave this decision to @sbourdeauducq

hartytp, 2018-08-30:
@sbourdeauducq why are we AC coupling the AD9154 sysref inputs? I guess it's not a problem so long as we are careful to gate the DAC's sensitivity to them, but it seems a bit odd not to DC couple this.

sbourdeauducq, 2018-08-31:
No good reason. Better DC-couple if possible (@gkasprow) as AC-coupling would interfere with the FF eye scan scheme I proposed.

sbourdeauducq, 2018-08-31:
It's not an ugly scheme at all. Probably it has less total transistors than the HMC7044 :) If chip vendors didn't suck, they would just put that circuit into one small low-cost device instead of inventing monstrosities.

sbourdeauducq, 2018-08-31:
> FWIW, my feeling is that the easiest way of doing this would probably be to replaced the Si5324 with a model with more outputs, such as a Si5366, which is essentially the same cost as the Si5324.

It's not the same, e.g. the loop bandwidth is different and there are other features. Maybe the simplest is to keep the 5324 (which has two outputs with independent dividers) and either keep the fanout buffer or modify siphaser so everything runs from the GT clock input and the fabric out is no longer required. I do not have time to look into this right now. When removing the fabric out, whoever does the test needs to check the impact on timing accuracy, as the GT input to the fabric path may have skew variations.

hartytp, 2018-09-01:
@gkasprow do you have a test board that breaks the IO from the high density connectors used on the AFEs out to something more accessible like 100mil header? If so, can you post me one when you send my Sayma back, please? This would be very useful for connecting the delay line eval board up to Sayma for synchronisation testing...

gkasprow, 2018-09-01:
Maybe you'd prefer to use clock mezzanine and its control lines? I have a board with all IO lines routed to I2C extender.

hartytp, 2018-09-02:
Thanks @gkasprow. However, I'm not sure that will help very much, as there is no convenient way of getting the IO off that test mezzanine. Ideally, I was hoping for a robust way of routing the IO off the board and into a wiring harness. e.g. a board with the high-density connector hooked up to 100mil header would be perfect. If that doesn't exist, I'll just use a PCB mount connector + some glue

hartytp, 2018-09-27:
@gkasprow  Are all the vias directly under the HMC7043 grounded?

gkasprow, 2018-09-27:
Yes. I verified it with multimeter to be 100% sure.

hartytp, 2018-09-28:
Here is an eye measurement.
- green is the 150MHz reference (another MCL power splitter)
- yellow is the phase-shifted sysref
- blue is the 2.4GHz DAC clock
- scope is on infinite persist with 10 min of DAQ time

As you can see, jitter isn't really an issue even for the max DAC clock

hartytp, 2018-09-28:
Oops, wrong eye photo, that one didn't have the clock on persist.  Should have been this one (still consistent with scope jitter)

[image]

hartytp, 2018-09-28:
Video showing the phase shifter working. Steppin one delay line in 1 LSB (11ps) increments at a rate of 1LSB/S. 

https://drive.google.com/open?id=1EJ4rNzYQF8aQdXBBAEc5-Vf_J85HfQrR

hartytp, 2018-09-28:
FWIW, heating the phase shifter and FF with an air gun that was hot enough to start melting my glue, I saw about 300ps pk-pk change in the phase at the scope. That suggests that once the board has warmed up for a few minutes the stability should be fine.

dhslichter, 2018-09-28:
@hartytp this is excellent!  Thanks for doing these tests, this seems like a nice solution for Sayma 2.0.

hartytp, 2018-09-28:
Yes, it all seems to work exactly how we want. Ideally, I'd like to see synchronized RF at the DAC's output on a scope, but the rework on Sayma required to do that is going to be a complete PITA, so I'm putting it off for now...

hartytp, 2018-09-28:
Comment about temperature coefficients: the delay line [data sheet](https://www.onsemi.com/pub/Collateral/NB6L295M-D.PDF) specifies a maximum PVT variation as 10.7ns-7.2ns=3.5ns. Assuming that's all due to temperature and that the variation is linear -- both probably quite pessimistic assumptions -- the temp co would be 25ps/K. The max frequency DAC clock is 2.4GHz, so 416ps period. For DC signals, S+H is 108+64ps. So, we need the phase to be good to about 200ps for this to work. That means that the phase shifter chip needs to be stable to about 8K, which should not be an issue at all.

Note, however, that we may need the DACs and phase shifters to warm up a little before synchronisation...

hartytp, 2018-09-29:
First go with Sayma RTM
- cut the trace connecting the HMC7043 to the AC coupling capacitors
- soldered coax onto both capacitor pads for mechanical stability
- AC couple the outputs of the phase shiftor and connect to the RTM via coax
- 50Ohm AC terminate the coax near the HMC7043 so there is a roughly 2inch stub after the termination. That's pretty nasty for a 2.4GHz signal! But, I don't have the tools to start soldering 0201 resistors across the DAC pads. 

- An eye scan on Sayma using the digital delay line
- taps are 11ps, 1/2.4GHz = 38taps, setup and hold are something like 10 taps each:

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1404.g05ae15be.dirty;standalone.without-sawg
Initializing SDRAM...
DQS initial delay: 110 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110011000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001101100000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100110110000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001101100000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 110 taps
Write leveling: 106 112 139 130 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100101010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000101101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 222+-84 212+-92 195+-86 180+-92 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000005s]  INFO(runtime): ARTIQ runtime starting...
[     0.003872s]  INFO(runtime): software ident 4.0.dev+1404.g05ae15be.dirty;standalone.without-sawg
[     0.012657s]  INFO(runtime): gateware ident 4.0.dev+1404.g05ae15be.dirty;standalone.without-sawg
[     0.021442s]  INFO(runtime): log level set to INFO by default
[     0.027156s]  INFO(runtime): UART log level set to INFO by default
[     0.033306s]  INFO(board_artiq::slave_fpga): Loading slave FPGA gateware...
[     0.040256s]  INFO(board_artiq::slave_fpga):   magic: 0x5352544d, length: 0x000c4a64
[... cut ...]
[   534.010160s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB
[   534.017431s]  WARN(runtime): using default MAC address 02-00-00-00-00-11; consider changing it
[   534.024818s]  INFO(runtime): using default IP address 192.168.1.60
[   534.032493s]  INFO(runtime::mgmt): management interface active
[   534.045670s]  INFO(runtime::session): accepting network sessions
[   534.059694s]  INFO(runtime::session): running startup kernel
[   534.089267s]  INFO(runtime::kern_hwreq): resetting RTIO

```

hartytp, 2018-09-29:
NB
- `true` indicates that a realignment has occurred (i.e. a sync error), so patches of `false` are where we want to work
- patches of true indicate a S/H violation. Assuming ~200ps S + H we expect the width of these patches to be <= 15 taps in the absence of any additional jitter and perfect rise times on our signal

So, we see some decent ~15tap=160ps wide working zones. That indicates that this should be (just about) workable.

In particular, based on this, my guess is that when we improve the SI a bit this will work really well

hartytp, 2018-09-29:
NB this is using the DAC's `one shot` mode, so the statistics are one measurement per point. Probably better to use one-shot then monitor mode in future to get better statistics

hartytp, 2018-09-29:
One thing to reiterate about this new synchronisation scheme: it only takes care of synchronising the SYSREF signal w.r.t. the local RTIO clock; it does not align the DAC clock itself. Instead, we rely on the HMC830 PLL having deterministic input (reference) to output (DAC) latency, which it does *so long as it's used in fundamental mode* (no output divider). That means it's limited to DAC clocks between 1.5GHz and 2GHz.

That't generally fine for us because we will want to use interpolation anyway. I've tested the 4x interpolation on Sayma with 600MHz clock rate and, pleasingly, it seems to work fine without any issues (just change one register value).

If it is an issue then I think we can get around it by routing a copy of the DAC clock to the RTM FPGA. We then recycle the SYSREF sampler logic that @sbourdeauducq has already implemented to measure the DAC clock phase w.r.t. the RTIO phase and reset the HMC830 (e.g. change the output divider between two values) until it has the correct phase. This should be relatively straightforwards to do (we already do something similar for the DRTIO transceivers), but someone has to implement it.

Anyway, I don't think any of this is a real issue for ARTIQ. @gkasprow can you comment on whether this is okay for non-ARTIQ users of Sayma.

hartytp, 2018-09-30:
@gkasprow what do you think about this:
- I'm doing eye scans for Sayma RTM DAC SYSREF.
- Taps are 11ps
- for each tap I take 11 points. '.' indicates that the DAC measured the same SYSREF phase as the previous attempt. 'x' means that it measured a different SYSREF phase and switched to the new phase (so `xx` usually means that it switched the phase twice, ending up at the original phase)
- specified DAC S/H timings are about 10taps

Typical eye scan for 600MHz DAC clock (x1 interpolation, 150 taps per clock cycle):
```
[    16.319847s]  INFO(board_artiq::jesd204sync): 0: x.........
[    16.434990s]  INFO(board_artiq::jesd204sync): 1: ..........
[    16.550130s]  INFO(board_artiq::jesd204sync): 2: ..........
[    16.665269s]  INFO(board_artiq::jesd204sync): 3: ..........
[    16.780409s]  INFO(board_artiq::jesd204sync): 4: ..........
[    16.895549s]  INFO(board_artiq::jesd204sync): 5: x.........
[    17.010689s]  INFO(board_artiq::jesd204sync): 6: ..........
[    17.125831s]  INFO(board_artiq::jesd204sync): 7: ..........
[    17.240971s]  INFO(board_artiq::jesd204sync): 8: ..........
[    17.356111s]  INFO(board_artiq::jesd204sync): 9: ..........
[    17.471251s]  INFO(board_artiq::jesd204sync): 10: ..........
[    17.586479s]  INFO(board_artiq::jesd204sync): 11: ..........
[    17.701706s]  INFO(board_artiq::jesd204sync): 12: ..........
[    17.816933s]  INFO(board_artiq::jesd204sync): 13: ..........
[    17.932161s]  INFO(board_artiq::jesd204sync): 14: ..........
[    18.047388s]  INFO(board_artiq::jesd204sync): 15: ..........
[    18.162618s]  INFO(board_artiq::jesd204sync): 16: ..........
[    18.277845s]  INFO(board_artiq::jesd204sync): 17: ..........
[    18.393073s]  INFO(board_artiq::jesd204sync): 18: ..........
[    18.508301s]  INFO(board_artiq::jesd204sync): 19: ..........
[    18.623528s]  INFO(board_artiq::jesd204sync): 20: ..........
[    18.738756s]  INFO(board_artiq::jesd204sync): 21: ..........
[    18.853984s]  INFO(board_artiq::jesd204sync): 22: ..........
[    18.969212s]  INFO(board_artiq::jesd204sync): 23: ..........
[    19.084439s]  INFO(board_artiq::jesd204sync): 24: ..........
[    19.199668s]  INFO(board_artiq::jesd204sync): 25: ..........
[    19.314896s]  INFO(board_artiq::jesd204sync): 26: ..........
[    19.430124s]  INFO(board_artiq::jesd204sync): 27: ..........
[    19.545352s]  INFO(board_artiq::jesd204sync): 28: ..........
[    19.660580s]  INFO(board_artiq::jesd204sync): 29: ..........
[    19.775808s]  INFO(board_artiq::jesd204sync): 30: ..........
[    19.891035s]  INFO(board_artiq::jesd204sync): 31: ..........
[    20.006263s]  INFO(board_artiq::jesd204sync): 32: ..........
[    20.121489s]  INFO(board_artiq::jesd204sync): 33: ..........
[    20.236717s]  INFO(board_artiq::jesd204sync): 34: ..........
[    20.351944s]  INFO(board_artiq::jesd204sync): 35: ..........
[    20.467172s]  INFO(board_artiq::jesd204sync): 36: ..........
[    20.582399s]  INFO(board_artiq::jesd204sync): 37: ..........
[    20.697626s]  INFO(board_artiq::jesd204sync): 38: ..........
[    20.812854s]  INFO(board_artiq::jesd204sync): 39: ..........
[    20.928082s]  INFO(board_artiq::jesd204sync): 40: ..........
[    21.043309s]  INFO(board_artiq::jesd204sync): 41: ..........
[    21.158539s]  INFO(board_artiq::jesd204sync): 42: ..........
[    21.273766s]  INFO(board_artiq::jesd204sync): 43: ..........
[    21.388994s]  INFO(board_artiq::jesd204sync): 44: ..........
[    21.504222s]  INFO(board_artiq::jesd204sync): 45: ..........
[    21.619450s]  INFO(board_artiq::jesd204sync): 46: ..........
[    21.734678s]  INFO(board_artiq::jesd204sync): 47: ..........
[    21.849905s]  INFO(board_artiq::jesd204sync): 48: ..........
[    21.965134s]  INFO(board_artiq::jesd204sync): 49: ..........
[    22.080362s]  INFO(board_artiq::jesd204sync): 50: ..........
[    22.195591s]  INFO(board_artiq::jesd204sync): 51: ..........
[    22.310819s]  INFO(board_artiq::jesd204sync): 52: ..........
[    22.426047s]  INFO(board_artiq::jesd204sync): 53: ..........
[    22.541275s]  INFO(board_artiq::jesd204sync): 54: ..........
[    22.656502s]  INFO(board_artiq::jesd204sync): 55: ..........
[    22.771730s]  INFO(board_artiq::jesd204sync): 56: ..........
[    22.886957s]  INFO(board_artiq::jesd204sync): 57: ..........
[    23.002185s]  INFO(board_artiq::jesd204sync): 58: ..........
[    23.117410s]  INFO(board_artiq::jesd204sync): 59: ..........
[    23.232638s]  INFO(board_artiq::jesd204sync): 60: ..........
[    23.347867s]  INFO(board_artiq::jesd204sync): 61: ..........
[... 1456 lines cut ...]
[    70.749710s]  INFO(board_artiq::jesd204sync): 488: xxxxxxxxxx
[    70.865037s]  INFO(board_artiq::jesd204sync): 489: xxxxxxxxxx
[    70.980364s]  INFO(board_artiq::jesd204sync): 490: xxxxxxxxxx
[    71.095692s]  INFO(board_artiq::jesd204sync): 491: x.......xx
[    71.211020s]  INFO(board_artiq::jesd204sync): 492: .......x.x
[    71.326347s]  INFO(board_artiq::jesd204sync): 493: ......x.x.
[    71.441674s]  INFO(board_artiq::jesd204sync): 494: ......xxxx
[    71.557001s]  INFO(board_artiq::jesd204sync): 495: xxxxxxxxxx
[    71.672330s]  INFO(board_artiq::jesd204sync): 496: xxxx.xxxxx
[    71.787657s]  INFO(board_artiq::jesd204sync): 497: xxx.x.....
[    71.902985s]  INFO(board_artiq::jesd204sync): 498: ..xxxxx...
[    72.018312s]  INFO(board_artiq::jesd204sync): 499: xx..xxxxxx
[    72.133642s]  INFO(board_artiq::jesd204sync): 500: .x........
[    72.248969s]  INFO(board_artiq::jesd204sync): 501: ..........
[... cut ...]
[    72.479624s]  INFO(board_artiq::jesd204sync): 503: .........x
[    72.594950s]  INFO(board_artiq::jesd204sync): 504: x.......xx
[    72.710277s]  INFO(board_artiq::jesd204sync): 505: x......x.x
[    72.825604s]  INFO(board_artiq::jesd204sync): 506: xxxxxxx.xx
[    72.940932s]  INFO(board_artiq::jesd204sync): 507: xxxxxxxxxx
[    73.056259s]  INFO(board_artiq::jesd204sync): 508: xxxxxxxx..
[    73.171588s]  INFO(board_artiq::jesd204sync): 509: ...xxxx...
[    73.286915s]  INFO(board_artiq::jesd204sync): 510: ....xx....

```

hartytp, 2018-09-30:
- my sysref is 9.375MHz in all cases, generated by the FPGA. It's retimed to the 150MHz reference clock that feeds the HMC830 and its phase is controlled by an external delay line. It's then fed to the DAC using external coax

hartytp, 2018-09-30:
So, @gkasprow the things I don't understand and want your input on are:
1. Why does it look great for 600MHz and so bad for higher DAC clock frequencies? If there is a SI issue shouldn't we see that on the 600MHz scan, but just with wider margins? The only explanation I can think of is that there is increased ground/etc noise at higher clock frequencies which makes the SI worse
2. There seems to be some periodic structure in the 1.2GHz sample. Note that each point for the same delay is a repeat measurement, so I would expect the 'x' and '.' to be random on each row.

hartytp, 2018-09-30:
All I can think of is that: I will have a go at doing a synchronization test at 600MHz (I will use a scope to post-select data points where the HMC830 divider starts with the correct phase since that's something we can't control without additional HW changes). Once I've done that, I'll send my whole test setup to Greg to see if he can improve the SI

hartytp, 2018-09-30:
Okay, I just don't understand that. Repeating the 1.2GHz measurement, but taking more points. 
The sequence here is:
- set delay line phase
- wait 100us 
- as the dac to check if the sysref phase has settled repeatedly

```
[    16.402040s]  INFO(board_artiq::jesd204sync): 0: x.....xx...........xx...........xx...........xx...........xx............xx......
[    16.598795s]  INFO(board_artiq::jesd204sync): 1: .......xx...........xx............xx...........xx...........xx...........xx.....
[    16.795522s]  INFO(board_artiq::jesd204sync): 2: .........xx...........xx...........xx...........xx............xx...........xx...
[    16.992250s]  INFO(board_artiq::jesd204sync): 3: ...........xx...........xx...........xx...........xx...........xx...........xxx.
[    17.188977s]  INFO(board_artiq::jesd204sync): 4: x...........xx...........xx............xx...........xx...........xx...........xx
[    17.385705s]  INFO(board_artiq::jesd204sync): 5: .xx...........xx...........xx...........xx............xx...........xx...........
[    17.582433s]  INFO(board_artiq::jesd204sync): 6: ...xx...........xx...........xx...........xx...........xx............xx.........
[    17.779160s]  INFO(board_artiq::jesd204sync): 7: ....xx...........xxx...........xx...........xx...........xx...........xx........
[    17.975888s]  INFO(board_artiq::jesd204sync): 8: ......xx...........xx...........xx............xx...........xx...........xx......
[    18.172616s]  INFO(board_artiq::jesd204sync): 9: ........xx...........xx...........xx...........xx............xx...........xx....
[    18.369342s]  INFO(board_artiq::jesd204sync): 10: ..........xx...........xx...........xx...........xx...........xx............xx..
[    18.566157s]  INFO(board_artiq::jesd204sync): 11: ...........xx...........xx............xx...........xx...........xx...........xx.
[    18.762973s]  INFO(board_artiq::jesd204sync): 12: xx...........xx...........xx...........xx...........xxx...........xx...........x
[    18.959788s]  INFO(board_artiq::jesd204sync): 13: .xx............xx...........xx...........xx...........xx...........xx...........
[    19.156605s]  INFO(board_artiq::jesd204sync): 14: ...xx...........xx...........xx............xx...........xx...........xx.........
[    19.353419s]  INFO(board_artiq::jesd204sync): 15: .....xx...........xx...........xx...........xx...........xxx...........xx.......
[    19.550234s]  INFO(board_artiq::jesd204sync): 16: ......xx............xx...........xx...........xx...........xx...........xx......
[    19.747049s]  INFO(board_artiq::jesd204sync): 17: ........xx...........xx...........xx............xx...........xx...........xx....
[    19.943864s]  INFO(board_artiq::jesd204sync): 18: ..........xx...........xx...........xx...........xx...........xxx...........xx..
[    20.140679s]  INFO(board_artiq::jesd204sync): 19: ...........xx............xx...........xx...........xx...........xx...........xx.
[    20.337493s]  INFO(board_artiq::jesd204sync): 20: xx...........xx...........xx............xx...........xx...........xx...........x
[    20.534309s]  INFO(board_artiq::jesd204sync): 21: ..xx...........xx...........xx...........xx............xx...........xx..........
[    20.731123s]  INFO(board_artiq::jesd204sync): 22: ...xx...........xxx...........xx...........xx...........xx...........xx.........
[    20.927939s]  INFO(board_artiq::jesd204sync): 23: .....xx...........xx...........xx............xx...........xx...........xx.......
[    21.124754s]  INFO(board_artiq::jesd204sync): 24: .......xx...........xx...........xx...........xx............xx...........xx.....
[    21.321570s]  INFO(board_artiq::jesd204sync): 25: .........xx...........xx...........xx...........xx...........xx............xx...
[    21.518386s]  INFO(board_artiq::jesd204sync): 26: ..........xx...........xxx...........xx...........xx...........xx...........xx..
[    21.715200s]  INFO(board_artiq::jesd204sync): 27: x...........xx...........xx...........xx............xx...........xx...........xx
[    21.912015s]  INFO(board_artiq::jesd204sync): 28: .xx...........xx...........xx...........xx...........xxx...........xx...........
[    22.108830s]  INFO(board_artiq::jesd204sync): 29: ...xx...........xx...........xx...........xx...........xx............xx.........
[    22.305645s]  INFO(board_artiq::jesd204sync): 30: ....xx...........xxx...........xx...........xx...........xx...........xx........
[    22.502461s]  INFO(board_artiq::jesd204sync): 31: ......xx...........xx...........xx...........xxx...........xx...........xx......
[    22.699275s]  INFO(board_artiq::jesd204sync): 32: ........xx...........xx...........xx...........xx...........xx............xx....
[    22.896090s]  INFO(board_artiq::jesd204sync): 33: .........xx...........xxx...........xx...........xx...........xx...........xx...
[    23.092906s]  INFO(board_artiq::jesd204sync): 34: ...........xx...........xx...........xx............xx...........xx...........xx.
[    23.289722s]  INFO(board_artiq::jesd204sync): 35: xx...........xx...........xx...........xx...........xx............xx...........x
[    23.486538s]  INFO(board_artiq::jesd204sync): 36: .xx............xx...........xx...........xx...........xx...........xx...........
[    23.683353s]  INFO(board_artiq::jesd204sync): 37: ...xx...........xx............xx...........xx...........xx...........xx.........
[    23.880169s]  INFO(board_artiq::jesd204sync): 38: .....xx...........xx...........xx............xx...........xx...........xx.......
[... cut ...]
[   110.123113s]  INFO(board_artiq::jesd204sync): 476: ........xx............xx...........xx............xx...........xx............xx..
[   110.320028s]  INFO(board_artiq::jesd204sync): 477: xx...........xxx...........xx...........xx............xx...........xx...........
[   110.516943s]  INFO(board_artiq::jesd204sync): 478: .....xx............xx...........xx............xx...........xx............xx.....
[   110.713860s]  INFO(
```

Pretty deterministically, about every 12th time we ask the DAC to check the SYSREF phase it gives us an error

hartytp, 2018-09-30:
Repeat of the same measurement, but with an extra 333us delay added between DAC phase checks. This shows that the odd pattern we see doesn't seem to be dependent on the gaps between phase checks. It really seems that the DAC just gives us a different answer sometimes.
```
[    16.385369s]  INFO(board_artiq::jesd204sync): 0: xxxxxxxxxxx..xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..x
[    16.582122s]  INFO(board_artiq::jesd204sync): 1: .xxxxxxxxxxx..xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..
[    16.778850s]  INFO(board_artiq::jesd204sync): 2: ..xxxxxxxxxxx..xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx.
[    16.975577s]  INFO(board_artiq::jesd204sync): 3: x..xxxxxxxxxxx..xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx
[    17.172304s]  INFO(board_artiq::jesd204sync): 4: xx..xxxxxxxxxxx..xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxx
[    17.369032s]  INFO(board_artiq::jesd204sync): 5: xxx..xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxx
[    17.565758s]  INFO(board_artiq::jesd204sync): 6: xxxx...xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxx
[    17.762486s]  INFO(board_artiq::jesd204sync): 7: xxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxx
[    17.959214s]  INFO(board_artiq::jesd204sync): 8: xxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx...xxxxx
[    18.155939s]  INFO(board_artiq::jesd204sync): 9: xxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxxx..xxxx
[    18.352668s]  INFO(board_artiq::jesd204sync): 10: xxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxxx..xxx
[    18.549483s]  INFO(board_artiq::jesd204sync): 11: xxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx...xx
[    18.746297s]  INFO(board_artiq::jesd204sync): 12: xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xx
[    18.943112s]  INFO(board_artiq::jesd204sync): 13: .xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..x
[    19.139927s]  INFO(board_artiq::jesd204sync): 14: .xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..
[    19.336744s]  INFO(board_artiq::jesd204sync): 15: ..xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx.
[    19.533559s]  INFO(board_artiq::jesd204sync): 16: x..xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxx
[    19.730373s]  INFO(board_artiq::jesd204sync): 17: xxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxx
[    19.927188s]  INFO(board_artiq::jesd204sync): 18: xxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxx
[    20.124003s]  INFO(board_artiq::jesd204sync): 19: xxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxx
[    20.320818s]  INFO(board_artiq::jesd204sync): 20: xxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxx
[    20.517634s]  INFO(board_artiq::jesd204sync): 21: xxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxxx.xxxxx
[    20.714449s]  INFO(board_artiq::jesd204sync): 22: xxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxx
[    20.911263s]  INFO(board_artiq::jesd204sync): 23: xxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxx
[    21.108077s]  INFO(board_artiq::jesd204sync): 24: xxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xx
[    21.304892s]  INFO(board_artiq::jesd204sync): 25: xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.x
[    21.501708s]  INFO(board_artiq::jesd204sync): 26: .xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.
[    21.698524s]  INFO(board_artiq::jesd204sync): 27: x.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx
[    21.895337s]  INFO(board_artiq::jesd204sync): 28: xx.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxx
[    22.092152s]  INFO(board_artiq::jesd204sync): 29: xxx.xxxxxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxx
[    22.288968s]  INFO(board_artiq::jesd204sync): 30: xxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxx
[    22.485784s]  INFO(board_artiq::jesd204sync): 31: xxxxx..xxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxx
[    22.682600s]  INFO(board_artiq::jesd204sync): 32: xxxxxx..xxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxx
[    22.879416s]  INFO(board_artiq::jesd204sync): 33: xxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxx
[    23.076232s]  INFO(board_artiq::jesd204sync): 34: xxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxx
[    23.273049s]  INFO(board_artiq::jesd204sync): 35: xxxxxxxxx.xxxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxx
[    23.469863s]  INFO(board_artiq::jesd204sync): 36: xxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxx
[    23.666677s]  INFO(board_artiq::jesd204sync): 37: xxxxxxxxxxx..xxxxxxxxxxx..xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xxxxxxxxxxxx.xx
[... cut ...]
[   109.909495s]  INFO(board_artiq::jesd204sync): 475: .........xx...........xx............xx...........xx...........xx............xx..
[   110.106409s]  INFO(board_artiq::jesd204sync): 476: x...........xxx...........xx...........xx............xx...........xx...........x
[   110.303323s]  INFO(board_artiq::jesd204sync): 477: ...xx...........xx...........xx............xx...........xx...........xxx........
[   110.500239s]  INFO(board_artiq::jesd204sync): 478: ......xx............xx...........xx...........xx............xx...........xx.....
[   110.697155s]  INFO(
```

hartytp, 2018-09-30:
Same but 2.4GHz DAC clock
```
[    16.436270s]  INFO(board_artiq::jesd204sync): 0: x.x.x.......x.x........xxxxxxxxxx.xxxxxxxxxxx........x.x.......x.x.......xxx....
[    16.659770s]  INFO(board_artiq::jesd204sync): 1: .xxxxx.xxxxxxxxxxx....x...x.x..x....x.x....x..x.x..xx...x.x......xx.xxxxxxxxx.xx
[    16.883245s]  INFO(board_artiq::jesd204sync): 2: xxxxxxxxxxx........x.x.......x.x.......x.x.......x.x........xx........xx........
[    17.106718s]  INFO(board_artiq::jesd204sync): 3: ..x.x.......x.x.......x.x........xx........xx........xx........x.x.......x.x....
[    17.330193s]  INFO(board_artiq::jesd204sync): 4: .....x.x........xx........xx........x.x.......x.x.......x.x.......x.x.......x.x.
[    17.553667s]  INFO(board_artiq::jesd204sync): 5: .........xx........x.x.......x.x.......x.x.......x.x.......x.x........xx........
[    17.777144s]  INFO(board_artiq::jesd204sync): 6: ..x.x.......x.x.......x.x.......x.x........xx........xx........x.x.......x.x....
[    18.000618s]  INFO(board_artiq::jesd204sync): 7: ......xx........xx........x.x.......x.x.......x.x.......x.x.......x.x.......x.x.
[    18.224092s]  INFO(board_artiq::jesd204sync): 8: xx.......x.x.......x.x.......x.x.......x.x........xx........xx........x.x.......
[    18.447567s]  INFO(board_artiq::jesd204sync): 9: ..x.x........xx........xx........xx........x.x.......x.x.......x.x.......x.x....
[    18.671037s]  INFO(board_artiq::jesd204sync): 10: ......x.x.......x.x.......x.x.......xxx.......x.x........xx........xx........xx.
[    18.894599s]  INFO(board_artiq::jesd204sync): 11: xx.......x.x.......x.x.......x.x........xx........xx........x.x.......xxx.......
[    19.118164s]  INFO(board_artiq::jesd204sync): 12: ..x.x........xx........xx........x.x.......x.x.......x.x.......x.x.......xxx....
[    19.341725s]  INFO(board_artiq::jesd204sync): 13: ......x.x.......x.x.......x.x.......x.x.......x.x........xx........xx........x.x
[    19.565288s]  INFO(board_artiq::jesd204sync): 14: xx.......xxx.......x.x........xx........xx........x.x.......xxx.......xxx.......
[    19.788850s]  INFO(board_artiq::jesd204sync): 15: ...xx........xx........xxxxxxxxxxx.x.......x.x.......x.x.......x.x........xx....
[    20.012413s]  INFO(board_artiq::jesd204sync): 16: ......x.x.......x.x.......x.x.......x.x........xx........xx........x.x.......x.x
[    20.235972s]  INFO(board_artiq::jesd204sync): 17: xx........xx.......xxx........xx........xxxxxxxxxxx.x.......xxx.......x.x.......
[    20.459533s]  INFO(board_artiq::jesd204sync): 18: ...xx........xxxxxxxxxxx.x.......x.x.......xxxxxxxxxxxxx........xx........xx....
[    20.683092s]  INFO(board_artiq::jesd204sync): 19: ......x.x.......x.x.......x.x........xx........xx........xxxxxxxxxxx.x.......x.x
[    20.906653s]  INFO(board_artiq::jesd204sync): 20: xx........xx........xx........x.x.......xxx.......x.x.......x.x.......x.x.......
[    21.130211s]  INFO(board_artiq::jesd204sync): 21: ...x.x.......x.x.......xxx.......xxx.......x.x........xx.......xxx........xx....
[    21.353772s]  INFO(board_artiq::jesd204sync): 22: ......x.x.......xxx........xx........xxxxxxxxxxx.x.......xxx.......x.x.......x.x
[    21.577330s]  INFO(board_artiq::jesd204sync): 23: xx........x.x.......xxx.......x.x.......xxx.......x.x.......x.x........xx.......
[    21.800891s]  INFO(board_artiq::jesd204sync): 24: ...x.x.......xxxxxxxxxxxxx........xx.......x.x........x.x.......x.x.......xxx...
[    22.024450s]  INFO(board_artiq::jesd204sync): 25: ......xxx........xxxxxxxxxxx.x.......xxx.......xxx.......xxx........xx........xx
[    22.248014s]  INFO(board_artiq::jesd204sync): 26: xxx.......xxxxxxxxxxx.x........xx........xx........x.x.......xxx.......xxx......
[    22.471577s]  INFO(board_artiq::jesd204sync): 27: ....xx........x.x.......x.x.......xxx.......x.x........xxxxxxxxxx.xxxxxxxxxxx...
[    22.695137s]  INFO(board_artiq::jesd204sync): 28: .......x.x........xxxxxxxxxxxx........x.x.......x.x.......xxx.......x.x.......x.
[    22.918698s]  INFO(board_artiq::jesd204sync): 29: xx.x.......x.x.......x.x.......xx.........xxxxxxxxxx.xxxxxxxxxxx........x.x.....
[    23.142261s]  INFO(board_artiq::jesd204sync): 30: .....xxxxxxxxxxxx........x.x.......x.x.......x.x.......xxxxxxxxxxx.xxxxxxxxx.xxx
[    23.365824s]  INFO(board_artiq::jesd204sync): 31: xxxxxxxxx.x........xxxxxxxxxx.xxxxxxxxxxxx.......xx........xxx.......xx........x
[    23.589383s]  INFO(board_artiq::jesd204sync): 32: x.x.x.......xx........x.x.......x.x........xxxxxxxxxx.xxxxxxxxxxx........x.x....
[    23.812945s]  INFO(board_artiq::jesd204sync): 33: ......xxxxxxxxxxxx........x.x.......xx........x.x.......x.x........xxxxxxxxxx.xx
[    24.036503s]  INFO(board_artiq::jesd204sync): 34: .xxxxxxxxx.x........xxxxxxxxxx.xxxxxxxxxxx........x.x.......x.x.......x.x.......
[    24.260070s]  INFO(board_artiq::jesd204sync): 35: ...x.x.......x.x.......x.x.......x.x........xxxxxxxxxxxx........x.x.......x.x...
[    24.483631s]  INFO(board_artiq::jesd204sync): 36: ......x.x........xx........x.x.......x.x.......x.x.......x.x........xxxxxxxxxx.x
[    24.707193s]  INFO(board_artiq::jesd204sync): 37: x.x.......x.x.......x.x........xxxxxxxxxxxx........x.x.......x.x.......x.x......
[    24.930755s]  INFO(board_artiq::jesd204sync): 38: ....x.x.......x.x.......x.x.......x.x........xxxxxxxxxxxx........x.x.......x.x..
[    25.154317s]  INFO(board_artiq::jesd204sync): 39: ........xx........x.x.......x.x.......x.x.......x.x........xxxxxxxxxxxx........x
[... cut ...]
[   122.664953s]  INFO(board_artiq::jesd204sync): 475: x.xxxxxxxxxxx.xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx.......xxxxxxxxxxxxxxxxxxxxxxxx...
[   122.888613s]  INFO(board_artiq::jesd204sync): 476: ........xxxxxxxxxxxxxxxxxxxxxxx........xxx.......x.x........xxxxxxxxxxxxx.......
[   123.112277s]  INFO(board_artiq::jesd204sync): 477: ....xxxxxxxxxxxxx.......xxx.......x.x........xxx.......x.x........xxxxxxxxxxx.x.
[   123.335939s]  INFO(board_artiq::jesd204sync): 478: xxxxxxxxxxxxx.......xxx.......x.x........xx........x.x.......x.x........xxxxxxxx
[   123.559603s]  INFO(
```

hartytp, 2018-09-30:
I'm convinced that this is not consistent with SI issues (how could we possibly get that kind of periodicity?). The obvious things I can think of are: (1) we're not doing the phase check correctly (@sbourdeauducq if I post my code this eve, can you do a review of the sync checks please?) (2) it's a hardware issue with the DAC at higher clock frequencies (worth contacting an ADI engineer?).

Either way, I think we need to resolve this before v2.0

hartytp, 2018-09-30:
eye scan in one-shot then monitor mode, reporting `    let sync_status = read(ad9154_reg::SYNC_CURRERR_L) as u32 + ((0x1 & read(ad9154_reg::SYNC_CURRERR_H) as u32) << 8);` now at 2.4GHz dac clock.

```
[    18.318132s]  INFO(board_artiq::jesd204sync): 0: 0
[    18.422824s]  INFO(board_artiq::jesd204sync): 1: 0
[    18.527513s]  INFO(board_artiq::jesd204sync): 2: 0
[    18.632203s]  INFO(board_artiq::jesd204sync): 3: 0
[    18.736892s]  INFO(board_artiq::jesd204sync): 4: 0
[    18.841582s]  INFO(board_artiq::jesd204sync): 5: 0
[    18.946271s]  INFO(board_artiq::jesd204sync): 6: 0
[    19.050961s]  INFO(board_artiq::jesd204sync): 7: 0
[    19.155652s]  INFO(board_artiq::jesd204sync): 8: 0
[    19.260342s]  INFO(board_artiq::jesd204sync): 9: 0
[    19.365031s]  INFO(board_artiq::jesd204sync): 10: 0
[    19.469808s]  INFO(board_artiq::jesd204sync): 11: 0
[    19.574584s]  INFO(board_artiq::jesd204sync): 12: 0
[    19.679361s]  INFO(board_artiq::jesd204sync): 13: 0
[    19.784138s]  INFO(board_artiq::jesd204sync): 14: 0
[    19.888916s]  INFO(board_artiq::jesd204sync): 15: 0
[    19.993693s]  INFO(board_artiq::jesd204sync): 16: 0
[    20.098470s]  INFO(board_artiq::jesd204sync): 17: 0
[    20.203248s]  INFO(board_artiq::jesd204sync): 18: 0
[    20.308025s]  INFO(board_artiq::jesd204sync): 19: 0
[    20.412802s]  INFO(board_artiq::jesd204sync): 20: 0
[    20.517579s]  INFO(board_artiq::jesd204sync): 21: 0
[    20.622356s]  INFO(board_artiq::jesd204sync): 22: 0
[    20.727133s]  INFO(board_artiq::jesd204sync): 23: 0
[    20.831909s]  INFO(board_artiq::jesd204sync): 24: 0
[    20.936686s]  INFO(board_artiq::jesd204sync): 25: 0
[    21.041463s]  INFO(board_artiq::jesd204sync): 26: 0
[    21.146241s]  INFO(board_artiq::jesd204sync): 27: 0
[    21.251019s]  INFO(board_artiq::jesd204sync): 28: 0
[    21.355795s]  INFO(board_artiq::jesd204sync): 29: 0
[    21.460572s]  INFO(board_artiq::jesd204sync): 30: 0
[    21.565349s]  INFO(board_artiq::jesd204sync): 31: 0
[    21.670126s]  INFO(board_artiq::jesd204sync): 32: 0
[    21.774902s]  INFO(board_artiq::jesd204sync): 33: 0
[    21.879679s]  INFO(board_artiq::jesd204sync): 34: 0
[    21.984456s]  INFO(board_artiq::jesd204sync): 35: 0
[    22.089233s]  INFO(board_artiq::jesd204sync): 36: 0
[    22.194011s]  INFO(board_artiq::jesd204sync): 37: 0
[    22.298788s]  INFO(board_artiq::jesd204sync): 38: 129
[    22.403740s]  INFO(board_artiq::jesd204sync): 39: 385
[    22.508692s]  INFO(board_artiq::jesd204sync): 40: 385
[    22.613644s]  INFO(board_artiq::jesd204sync): 41: 385
[    22.718597s]  INFO(board_artiq::jesd204sync): 42: 385
[    22.823549s]  INFO(board_artiq::jesd204sync): 43: 385
[    22.928501s]  INFO(board_artiq::jesd204sync): 44: 385
[    23.033453s]  INFO(board_artiq::jesd204sync): 45: 385
[    23.138406s]  INFO(board_artiq::jesd204sync): 46: 385
[    23.243358s]  INFO(board_artiq::jesd204sync): 47: 385
[    23.348311s]  INFO(board_artiq::jesd204sync): 48: 385
[    23.453263s]  INFO(board_artiq::jesd204sync): 49: 385
[    23.558215s]  INFO(board_artiq::jesd204sync): 50: 385
[    23.663167s]  INFO(board_artiq::jesd204sync): 51: 385
[    23.768119s]  INFO(board_artiq::jesd204sync): 52: 385
[    23.873071s]  INFO(board_artiq::jesd204sync): 53: 385
[    23.978024s]  INFO(board_artiq::jesd204sync): 54: 432
[    24.082976s]  INFO(board_artiq::jesd204sync): 55: 385
[    24.187930s]  INFO(board_artiq::jesd204sync): 56: 385
[    24.292882s]  INFO(board_artiq::jesd204sync): 57: 385
[    24.397834s]  INFO(board_artiq::jesd204sync): 58: 385
[    24.502786s]  INFO(board_artiq::jesd204sync): 59: 385
[    24.607738s]  INFO(board_artiq::jesd204sync): 60: 385
[    24.712690s]  INFO(board_artiq::jesd204sync): 61: 385
[    24.817642s]  INFO(board_artiq::jesd204sync): 62: 385
[    24.922594s]  INFO(board_artiq::jesd204sync): 63: 385
[    25.027547s]  INFO(board_artiq::jesd204sync): 64: 385
[    25.132500s]  INFO(board_artiq::jesd204sync): 65: 385
[    25.237453s]  INFO(board_artiq::jesd204sync): 66: 385
[... 420 lines cut ...]
[    69.460590s]  INFO(board_artiq::jesd204sync): 487: 396
[    69.565643s]  INFO(board_artiq::jesd204sync): 488: 448
[    69.670696s]  INFO(board_artiq::jesd204sync): 489: 396
[    69.775749s]  INFO(board_artiq::jesd204sync): 490: 396
[    69.880802s]  INFO(board_artiq::jesd204sync): 491: 396
[    69.985855s]  INFO(board_artiq::jesd204sync): 492: 444
[    70.090907s]  INFO(board_artiq::jesd204sync): 493: 444
[    70.195962s]  INFO(board_artiq::jesd204sync): 494: 444
[    70.301015s]  INFO(board_artiq::jesd204sync): 495: 444
[    70.406068s]  INFO(board_artiq::jesd204sync): 496: 396
[    70.511121s]  INFO(board_artiq::jesd204sync): 497: 396
[    70.616174s]  INFO(board_artiq::jesd204sync): 498: 396
[    70.721227s]  INFO(board_artiq::jesd204sync): 499: 444
[    70.826280s]  INFO(board_artiq::jesd204sync): 500: 396
[    70.931333s]  INFO(board_artiq::jesd204sync): 501: 396
[    71.036386s]  INFO(board_artiq::jesd204sync): 502: 396
[    71.141440s]  INFO(board_artiq::jesd204sync): 503: 444
[    71.246494s]  INFO(board_artiq::jesd204sync): 504: 444
[    71.351547s]  INFO(board_artiq::jesd204sync): 505: 444
[    71.456600s]  INFO(board_artiq::jesd204sync): 506: 396
[    71.561653s]  INFO(board_artiq::jesd204sync): 507: 396
[    71.666706s]  INFO(board_artiq::jesd204sync): 508: 444
[    71.771759s]  INFO(board_artiq::jesd204sync): 509: 444
[    71.876812s]  INFO(board_artiq::jesd204sync): 510: 396
```

sbourdeauducq, 2018-10-01:
Will start working on this on Wednesday.

hartytp, 2018-10-01:
Thanks @sbourdeauducq ! I'm becoming more convinced that this is either a bug in the AD9154 or us misunderstanding how the synchronization logic works (the documentation is a bit sparse and ambiguous). Trying a different procedure:
- init the SYSREF to a calibrated value. For now I'm using 255
- put the AD9154 in `one-shot then monitor` mode and enable the SYSREF machine
- sweep the delay line phase while measuring the phase error `currerr = currerr_l + ((currerr_h)&0x01) << 8)`

Here is an eye scan doing that, taking 10 data points at each delay line phase. Apologies for the ugly printing.

This is using the 1.2 GHz DAC clock, so approx 76 11ps taps per clock cycle.

[1G2_eye.txt](https://github.com/sinara-hw/sinara/files/2433478/1G2_eye.txt)

- The working region extends from 200 to 281 with a nice sharp transition edge and no dead zones in the middle. So that looks good!
- I haven't got my head around what those numbers actually mean. Why is the counter value so large? I didn't think the LMFC got that high
- the synchronisation doesn't seem to be 100% reliable. Sometimes I get a wall of 500ish numbers in the region where the edge should be. So, I think our process will need to be something like:
  - set the delay to the calibrated value
  - enable one-shot then monitor mode and arm sync machine
  - do an eye scan. if we find a working region where we expect it then check the margins, otherwise rearm the machine and go again

anyway, at least this method gets rid of that odd periodic behaviour

hartytp, 2018-10-01:
And, here is a synchronized DAC at 2.4GHz. 

[eyes2.txt](https://github.com/sinara-hw/sinara/files/2433619/eyes2.txt)

Working region is 38 taps which is spot on what we expect.

The bad news? It took a few power cycles of the device to produce this. Provisionally, then, it seems that at the higher clock frequencies the sync logic does not always work. But once the device is synchronized it stays synchronized fairly well (will need to do a longer term measurement to make sure) and gives good eye scans.

Next step: see if we can achieve good eye scans on every boot by repeatedly re-arming the sync machine until it works.

hartytp, 2018-10-01:
```rust
fn sysref_cal_dac(dacno: u8) -> Result<u16, &'static str> {
    info!("calibrating SYSREF phase at DAC-{}...", dacno);

    let mut d = 255 as u16;
    let mut dmin = d;
    let mut dmax = d;

    for _ in 0..100 {
        info!("checking DAC {} SYSREF phase...", dacno);

        dmin = d;
        dmax = d;
        delay_line::set_delay(0, d);
        delay_line::set_delay(1, d);
        clock::spin_us(10000);
        ad9154::dac_arm_sync(dacno);

        // measure negative margin
        loop {
            delay_line::set_delay(0, dmin);
            delay_line::set_delay(1, dmin);
            clock::spin_us(1000);

            let mut synchronised = true;
            for check in 0..500 {
                if ad9154::check_dac_sync(dacno) != 0 {
                    synchronised = false;
                    break;
                }
            }
            if !synchronised { break; }
            dmin -= 1;
        }

        // measure positive margin
        loop {
            delay_line::set_delay(0, dmax);
            delay_line::set_delay(1, dmax);
            clock::spin_us(1000);

            let mut synchronised = true;
            for check in 0..500 {
                if ad9154::check_dac_sync(dacno) != 0 {
                    synchronised = false;
                    break;
                }
            }
            if !synchronised { break; }
            dmax += 1;
        }

        info!("dmin {}, dmax {}", dmin, dmax);
        if dmax - dmin > 10 {
            delay_line::set_delay(dacno, d);
            break;
        }
    }
    if dmax - dmin > 10 { Ok(d) }
    else { Err("failed to align sysref") }
}
```

```rust

pub fn dac_arm_sync(dacno: u8) {
    spi_setup(dacno);
    write(ad9154_reg::SYNC_CONTROL,
        0x9*ad9154_reg::SYNCMODE | 0*ad9154_reg::SYNCENABLE);
    clock::spin_us(1000);
    write(ad9154_reg::SYNC_CONTROL,
        0x9*ad9154_reg::SYNCMODE | 1*ad9154_reg::SYNCENABLE);
    clock::spin_us(1000);
    write(ad9154_reg::SYNC_CONTROL,
        0x9*ad9154_reg::SYNCMODE | 1*ad9154_reg::SYNCENABLE |
        1*ad9154_reg::SYNCARM);
}

pub fn check_dac_sync(dacno: u8) -> u32 {
    spi_setup(dacno);

    let err_l = read(ad9154_reg::SYNC_CURRERR_L) as u32;
    let err_h = 0x1 & read(ad9154_reg::SYNC_CURRERR_H) as u32;
    return err_l + (err_h << 8)
}
```
Seems to work.

I now get power on results like:
```
[    16.178015s]  INFO(board_artiq::jesd204sync): calibrating SYSREF phase at DAC-1...
[    16.184278s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.205899s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.210590s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.232207s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.236896s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.258513s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.263203s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.284821s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.289510s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.311127s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.315816s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.337433s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.342123s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.363740s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.368430s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.390047s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.394736s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.416353s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.421042s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.442659s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.447349s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    16.468967s]  INFO(board_artiq::jesd204sync): dmin 255, dmax 255
[    16.473656s]  INFO(board_artiq::jesd204sync): checking DAC 1 SYSREF phase...
[    18.485236s]  INFO(board_artiq::jesd204sync): dmin 241, dmax 280
```

Let's see if that actually translates into synchronised RF...

hartytp, 2018-10-02:
[image]

closed 2018-10-02 by hartytp

hartytp, 2018-10-02:
Detailed setup:
- Sayma standalone clocked at 150MHz from a synth (call this `ref_clk`)
- HMC830 PLL produces 2.4GHz `dac_clk` (deterministic latency w.r.t. `ref_clk` since output divider not used)
- HMC7043 DAC clock outputs in "fundamental mode" i.e. no dividers so still deterministic latency
- HMC7043 clocks the FPGA at 150MHz this uses a divider, so gives the 150MHz `rtio_clk` a random phase w.r.t the `ref_clk`
- FPGA produces a `sysref` signal at 9.375MHz. This is used to reset the FPGA JESD core
- FPGA SYSREF signal output on SMA 0
- FPGA SYSREF is retimed using an external FF that's clocked from `ref_clk`. NB I do not bother to control the phase of the FPGA SYSREF output w.r.t. the `ref_clk`. I just hope that it meets S/H, which it generally does, but for some restarts this isn't true.
- retimed SYSREF passes through a digital delay line
- delay is set to ensure that SYSREF meets S/H w.r.t. `dac_clk` at the DAC pin
- The code below is a cleaned up version of the experiment I'm running on the FPGA as a startup Kernel. The idea is that this switches the SAWG on and off at the same time (up to a constant latency) as the SYSREF edges, clearing the phase accumulator for each run of the experiment
- the screenshot above was taken with infinte persist on, accross ~10 power cycles of Sayma
  - Yellow trace is a DAC output
  - Green trace is `ref_clk`
  - blue trace is the phase-shifted SYSREF
- We see that the timing between all clocks is fixed (no 416ps indeterminism) indicating that the RF output from Sayma is correctly synchronised with the FPGA-generated SYSREF
- additionally, I have verified that the *un-retmined* FPGA-generated SYSREF is synchronised with local TTL events by running the same experiment and looking at the two FPGA SMA outputs (one of which is SYSREF and one of with is controlled by a kernel). Note that the kernel-controlled TTL is not completely synchronised with the RF because of the way I clock it. If I clocked the FPGA from the `ref_clk` directly this wouldn't be a problem, but I'd have to mess around with gateware to do that and I don't have time right now

hartytp, 2018-10-02:
tl;dr since the above shows that the RF is synchronised with local RTIO events, and we know that DRTIO correctly synchronises RTIO events between devices, I believe that the above is a complete demonstration of board-board synchronisation on sayma

hartytp, 2018-10-02:
Looking over a further 10 power cycles of the device, I did see 2 times when it did not correctly synchronize: 

[image]

Best guess is that this is due to the FPGA SYSREF not meeting S/H at the retiming FF

hartytp, 2018-10-02:
Are there any other measurements that people want to see before I disassemble my test setup and we finalize our plans for Sayma v2.0? (@sbourdeauducq @jordens @gkasprow )

sbourdeauducq, 2018-10-02:
Hmm I suggest investigating why it sometimes does not synchronize. Can you change the timing of the FPGA output that goes to the FF?

hartytp, 2018-10-02:
@sbourdeauducq for the sake of completeness, I redid the clocking on my Sayma. Now:
- AMC is clocked at 150MHz using a power splitter to take power from `ref_clk` and a balun to drive the SYSCLK1 input
- I'm now triggering my measurement from the kernel-controlled TTL and looking at the delay between the TTL and the RF output
- about 60%-80% of the time (not sure of the exact statistics), the latency is completely deterministic.
- However, some times there jumps in the TTL->RF latency. The size of these jumps looks to be 1/600MHz=1.66ns which is the data rate
- Given that the jumps are not spaced by 1/2.4GHz, we can conclude that the DAC *is* reliably synchronising to the SYSREF (the RF is always aligned to a 150MHz edge), but that there is some non-determinism somewhere else. e.g. a bug in my code or in the JESD core

Will post a photo in a sec

hartytp, 2018-10-02:
@sbourdeauducq based on the symptoms I'm seeing, I do not believe that this issue can be caused by S/H violations at that FF. I think it can only really be some issue to do with the gateware i.e. a bug in my code or a bug in the JESD core or somewhere in between

hartytp, 2018-10-02:
@sbourdeauducq one thing I did wonder about was the CDC between the reset of the JESD core and the JESD transceiver clock. Right now, the JESD core is clocked from the HMC7043 divider output, but the AMC FPGA fabric is clocked from my external 150MHz reference. Since there is no well-defined phase relationship between these two, I wonder if the CDC can add a cycle or two of non-determinism? That type of issue should be fixed in v2.0 when we have more suitable clocking options.

Anyway, if you do a review of my code and spot anything else you want me to try out then let me know. Otherwise, I'm happy to leave this here for now as I do feel that the point has been proved even if there is still some gateware issue to fix

sbourdeauducq, 2018-10-02:
>  I do feel that the point has been proved even if there is still some gateware issue to fix

Not really; >99% of Sayma development time is spent banging one's head on the desk wondering why it sometimes does not work.

hartytp, 2018-10-02:
@sbourdeauducq sure. But, I'm trying to triage this. My goal in this issue is to prove that with this hardware the DACs can correctly be synchronised at full speed. Based on the data we have, I believe that is the case (if the DACs weren't picking up the SYSREF edges correctly, the jitter would not be quantized in 1/600MHz steps).

Getting rid of the gateware bugs to make sure that the RTIO clock is synchronized with the RF is a separate issue IMHO. And, it's one that I don't feel qualified to deal with -- I don't understand the JESD core or the overall data flow in the gateware enough to contribute to that.

[6 more replies]