sinara-hw/sinara #584: Sayma: Remove CAL ADC

hartytp, 2018-08-30:
While we're thinking about changes to Sayma, does anyone still *really* want the CAL ADC? Or, can we scrap it?

gkasprow, 2018-08-30:
It can be added when needed to the AFE...

hartytp, 2018-08-30:
Good point. That might be a better place for it anyway (means that it's easy to fix any issues we find without having to respin Sayma RTM).

dhslichter, 2018-08-30:
Agreed that the AFE is the place for it.  However, I would like to have it available as an option still, so we should plan for it on the AFE.  It can be I2C on the AFE, honestly, super simple to implement.

hartytp, 2018-08-30:
@dhslichter 
- Let's definitely make sure that the AFE cards have plenty of GPIO from the AMC + RTM FPGAs. That way, users can put whatever ADCs + other ICs they like on the AFEs. The AFEs are simple, open-source boards that are relatively quick + cheap to respin (particularly if we don't over spec the substrate) so very easy for users to customize according to their needs
- I anticipate that users will all have their own pet features that they want to include on the AFEs. If we try to cram everyone's favorite bells and whistles on BaseMod/MixMod then we will end up with another over complicated mess. So, please can we all try to keep those designs to just core functionality, on the grounds that users are welcome to customize them according to their needs
- at this point, I've lost track of what the 24-bit AFC was for. We already agreed to scrap the power detector on the AFE, so what is it digitizing anyway?
- Having said all that, if you really think this is an important feature, can you open an issue on the BaseMod/MixMod projects, please describing exactly which ADC you want, what you want it connected to, etc. (although, we should probably hold off an in-depth discussion of this until later as I think that everyone is too busy to discuss this properly atm)

dhslichter, 2018-08-31:
I had not realized the power detector had been scrapped; the idea for this ADC would be to monitor power drifts of the AFE output, for example.  Anyway, I agree that this all should go in the BaseMod/MixMod discussion, and the point here is that we don't want it on the Sayma RTM board proper.

hartytp, 2018-09-23:
@dhslichter anyway, if you do really want to keep the power detector and the ADC, I still don't think there is much point using a 24-bit one as we currently do. A cheap I2C AC on the AFE would do the job just fine...

jbqubit, 2018-10-04:
> the idea for this ADC would be to monitor power drifts of the AFE output, for example.

AFAIR the rationale for scrapping the power detector is that external power amplifier drift is expected to dominate. Adding ADC on AFE is just a variation on the same theme.

hartytp, 2018-10-04:
@jbqubit There are two decoupled issues here:
1. Do we want a 24-bit multi-channel ADC on Sayma RTM
2. Do we want a power detector on the AFE

My argument is that (1) we do not want a 24-bit ADC on Sayma RTM. I don't think it's that useful (it's very slow and multiplexed between the channels which is annoying) and I do not think that a clear use case for it has been articulated. *if* we need ADCs to read out signals from the AFE mezzanines, then I think it'd be better to put an ADC on the AFE and route digital signals, rather than routing analog signals around. This also allows us to match the bandwidth/noise of the ADC to the application (e.g. a faster, lower-resolution ADC would be more appropriate for a power detector).

(2) for the reasons articulated previously, I still don't think it's particularly useful to have an RF power detector on the AFE. If you want leveling then place a power detector (can be a thumb-nail sized PCB) near the ion trap and connect that to the MSPS ADC on the AFE to do proper in-loop control of the RF amplitude. However, it's not something that I feel that strongly about and I don't want to have a lengthly discussion about a relatively unimportant point; paint the bike sheds fluorescent green if you must, just get the damn thing built!

gkasprow, 2018-11-27:
removed.

closed 2018-11-27 by gkasprow