sinara-hw/sinara #585: bad CODEGRPSYNCFLG

hartytp, 2018-09-20:
```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1400.g69d060b6;standalone.without-sawg
Initializing SDRAM...
DQS initial delay: 110 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101100000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101101000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011011000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011010000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 110 taps
Write leveling: 102 112 139 128 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110010010100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 223+-85 211+-91 195+-85 180+-91 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000006s]  INFO(runtime): ARTIQ runtime starting...
[     0.003867s]  INFO(runtime): software ident 4.0.dev+1400.g69d060b6;standalone.without-sawg
[     0.012130s]  INFO(runtime): gateware ident 4.0.dev+1400.g69d060b6;standalone.without-sawg
[     0.020771s]  INFO(runtime): log level set to INFO by default
[     0.026297s]  INFO(runtime): UART log level set to INFO by default
[     0.032260s]  INFO(board_artiq::slave_fpga): Loading slave FPGA gateware...
[     0.039210s]  INFO(board_artiq::slave_fpga):   magic: 0x5352544d, length: 0x000b5ce8
[     0.046937s]  INFO(board_artiq::slave_fpga):   DONE before loading
[     0.999431s]  INFO(board_artiq::slave_fpga):   ...done
[     1.003243s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     1.017001s]  INFO(board_artiq::serwb):  ...done.
[     1.020374s]  INFO(board_artiq::serwb): RTM to AMC link test...
[     2.502676s]  INFO(board_artiq::serwb):   ...passed
[     2.506222s]  INFO(board_artiq::serwb): AMC to RTM link test...
[     3.988532s]  INFO(board_artiq::serwb):   ...passed
[     3.992088s]  INFO(board_artiq::serwb): Wishbone test...
[     5.966622s]  INFO(board_artiq::serwb):   ...passed
[     5.970425s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+1400.g69d060b6
[     5.977811s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     6.977006s]  INFO(runtime): continuing boot
[     7.239248s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[    13.538077s]  INFO(board_artiq::si5324):   ...locked
[    13.541849s]  INFO(board_artiq::hmc830_7043::hmc830): loading HMC830 configuration...
[    13.549790s]  INFO(board_artiq::hmc830_7043::hmc830):   ...done
[    13.555435s]  INFO(board_artiq::hmc830_7043::hmc830): setting HMC830 dividers...
[    13.562902s]  INFO(board_artiq::hmc830_7043::hmc830):   ...done
[    13.568720s]  INFO(board_artiq::hmc830_7043::hmc830): waiting for HMC830 lock...
[    13.576141s]  INFO(board_artiq::hmc830_7043::hmc830):   ...locked
[    13.582374s]  INFO(board_artiq::hmc830_7043::hmc7043): enabling HMC7043
[    13.599078s]  INFO(board_artiq::hmc830_7043::hmc7043): loading configuration...
[... 65 lines cut ...]
[    15.796536s]  INFO(board_artiq::jesd204sync): calibrating SYSREF phase offset at FPGA...
[    15.813857s]  INFO(board_artiq::jesd204sync):   ...done, phase offset: 137
[    15.824821s]  INFO(board_artiq::jesd204sync): aligning SYSREF with RTIO...
[    15.851043s]  INFO(board_artiq::jesd204sync):   ...done (64/64 slips)
[    15.859210s]  INFO(board_artiq::jesd204sync):   margins at FPGA: -17 +17
[    15.864981s]  INFO(board_artiq::jesd204sync): verifying SYSREF margins at DAC-0...
[    15.962524s]  INFO(board_artiq::jesd204sync):   margins: -0 +69
[    15.967125s] ERROR(runtime): failed to align SYSREF at DAC: SYSREF margins at DAC are too small, board needs recalibration
[    15.978156s]  INFO(board_artiq::hmc542): card 0 channel 0 set to 4 dB
[    15.986675s]  INFO(board_artiq::hmc542): card 0 channel 1 set to 4 dB
[    15.993899s]  INFO(board_artiq::hmc542): card 1 channel 0 set to 4 dB
[    16.001122s]  INFO(board_artiq::hmc542): card 1 channel 1 set to 4 dB
[    16.008346s]  INFO(board_artiq::hmc542): card 2 channel 0 set to 4 dB
[    16.015569s]  INFO(board_artiq::hmc542): card 2 channel 1 set to 4 dB
[    16.022796s]  INFO(board_artiq::hmc542): card 3 channel 0 set to 4 dB
[    16.030022s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB
[    16.037479s]  INFO(runtime): using MAC address 34-45-32-12-ff-22
[    16.042437s]  INFO(runtime): using IP address 192.168.95.176
[    16.049459s]  INFO(runtime::mgmt): management interface active
[    16.062605s]  INFO(runtime::session): accepting network sessions
[    16.076550s]  INFO(runtime::session): running startup kernel
[    16.081287s]  INFO(runtime::session): no startup kernel found
[    16.086724s]  INFO(runtime::session): no connection, starting idle kernel
[    16.093870s]  INFO(runtime::session): no idle kernel found
```

hartytp, 2018-09-20:
@marmeladapk did you test the AMC + RTM you sent me? If so, please could you send me the binaries you used?

hartytp, 2018-09-20:
Anyway, this is currently stopping me from looking at Sayma synchronization...

gkasprow, 2018-09-20:
The cards we shipped to you and Sebastien were the two the best behaving. No issues with DAC config.
@marmeladapk is on a way to USA right now.

hartytp, 2018-09-21:
Thanks Greg.

@marmeladapk when you have email access again, please can you tell me which tests you did on this board and which binaries you used. I'll reproduce your tests and see if I can get the board to work.

@gkasprow do you still have my AMC + RTM? My RTM is identifiable because I removed the screening cans. My AMC is identifiable because I put a solder blob acress the jumper to fix the JTAG issues. If you can find those two boards can you ship them back to me so I can test synchronisation. 

All we need to do for Sayma v2.0 is the synchronisation tests and to check the SI for the new AFE test board you made.

gkasprow, 2018-09-21:
@marmeladapk took all the RTMs to USA. I have only single AMC in the lab.

hartytp, 2018-09-21:
:( @jbqubit @marmeladapk if you can find my RTM + AMC in the batch that @marmeladapk took with him then please can you post it to me asap. Until I have a working board I cannot test the new synchronisation scheme.

hartytp, 2018-09-21:
@gkasprow the P2V0 LD4 and P1V2 LD7 LEDs are off on this RTM

gkasprow, 2018-09-21:
the transistor nearby was probably broken mechanically during insertion to the  crate.

hartytp, 2018-09-21:
Okay, yes, someone has smashed a transistor near those LEDs (my board was never put into a crate, so didn't have this particular issue).

Does that affect the supplies or just the LEDs?

gkasprow, 2018-09-21:
just LEDs. It happened on many boards.
We shipped you one of the two boards that fully worked and didn't have any sync issues.

marmeladapk, 2018-09-22:
@hartytp AMC should have a sticker with rxcclock phase. Please tell me this value.

Did you test the synchronization with the gateware and firmware that were flashed on the board when you got it?

hartytp, 2018-09-22:
> AMC should have a sticker with rxcclock phase. Please tell me this value.

I don't remember seeing that, but I'll have another look next time I'm in the lab.

> Did you test the synchronization with the gateware and firmware that were flashed on the board when you got it?

No, I didn't. Can you send me the binaries you used and I'll re-test. Also, let me know what your setup (e.g. clocks etc)

Did you manage to track down where my AMC + RTM? I've used the same boards for all testing so far, and did some rework on them myself, so I'd really like to finish testing on that same pair of boards instead of switching to some other boards, whose history I know nothing about.

hartytp, 2018-09-22:
No red stickers. Did you post me the wrong board? To help you track down my AMC, IIRC it was missing one of the thumbscres on the FP.
[image]
[image]

hartytp, 2018-09-22:
Thanks, I'll give that a spin on Monday. Will let you know how I get on.

hartytp, 2018-09-23:
Thanks @marmeladapk. Strange, with the binaries you sent me, this works fine:
```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1335.gfab6e5cd.dirty;standalone
Initializing SDRAM...
DQS initial delay: 110 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 110 taps
Write leveling: 102 112 138 128 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000101010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100101001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010001010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000101011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101010100010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010100100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 222+-84 212+-93 191+-84 178+-92 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000005s]  INFO(runtime): ARTIQ runtime starting...
[     0.003867s]  INFO(runtime): software ident 4.0.dev+1359.g0e741945.dirty;standalone
[     0.011521s]  INFO(runtime): gateware ident 4.0.dev+1335.gfab6e5cd.dirty;standalone
[     0.019525s]  INFO(runtime): log level set to INFO by default
[     0.025085s]  INFO(runtime): UART log level set to INFO by default
[     0.031044s]  INFO(board_artiq::slave_fpga): Loading slave FPGA gateware...
[     0.037994s]  INFO(board_artiq::slave_fpga):   magic: 0x5352544d, length: 0x000b4c90
[     0.986799s]  INFO(board_artiq::slave_fpga):   ...done
[     0.990609s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     1.025338s]  INFO(board_artiq::serwb):  ...done.
[     1.028708s]  INFO(board_artiq::serwb): RTM to AMC link test...
[     2.511011s]  INFO(board_artiq::serwb):   ...passed
[     2.514556s]  INFO(board_artiq::serwb): AMC to RTM link test...
[     3.996865s]  INFO(board_artiq::serwb):   ...passed
[     4.000418s]  INFO(board_artiq::serwb): Wishbone test...
[... cut ...]
[    16.368270s]  INFO(runtime::session): accepting network sessions
[    16.382221s]  INFO(runtime::session): running startup kernel
[    16.386962s]  INFO(runtime::session): no startup kernel found
[    16.392395s]  INFO(runtime::session): no connection, starting idle kernel
[    16.399537s]  INFO(runtime::session): no idle kernel found

```

@sbourdeauducq can you think of any changes since that build that could account for the issues I'm seeing? (Particularly anything related to the clocking?)

hartytp, 2018-09-23:
For good measure, I rebuilt the current artiq master in a fresh conda environment. I'm actually mainly seeing `bad SYNC` errors. Any ideas which change to artiq could be responsible for this? 
```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1400.g69d060b6;standalone.without-sawg
Initializing SDRAM...
DQS initial delay: 110 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110110010000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101101000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 110 taps
Write leveling: 102 112 139 128 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110100010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101010100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000101011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 222+-85 211+-92 193+-83 179+-91 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000006s]  INFO(runtime): ARTIQ runtime starting...
[     0.003867s]  INFO(runtime): software ident 4.0.dev+1400.g69d060b6;standalone.without-sawg
[     0.012130s]  INFO(runtime): gateware ident 4.0.dev+1400.g69d060b6;standalone.without-sawg
[     0.020778s]  INFO(runtime): log level set to INFO by default
[     0.026302s]  INFO(runtime): UART log level set to INFO by default
[     0.032260s]  INFO(board_artiq::slave_fpga): Loading slave FPGA gateware...
[     0.039210s]  INFO(board_artiq::slave_fpga):   magic: 0x5352544d, length: 0x000b5ce8
[     0.046936s]  INFO(board_artiq::slave_fpga):   DONE before loading
[     0.999431s]  INFO(board_artiq::slave_fpga):   ...done
[     1.003243s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     1.037973s]  INFO(board_artiq::serwb):  ...done.
[     1.041346s]  INFO(board_artiq::serwb): RTM to AMC link test...
[     2.523648s]  INFO(board_artiq::serwb):   ...passed
[... cut ...]
[    15.933525s]  INFO(runtime): using MAC address 34-45-32-12-ff-22
[    15.938493s]  INFO(runtime): using IP address 192.168.95.176
[    15.945599s]  INFO(runtime::mgmt): management interface active
[    15.958751s]  INFO(runtime::session): accepting network sessions
[    15.972694s]  INFO(runtime::session): running startup kernel
[    15.977428s]  INFO(runtime::session): no startup kernel found
[    15.982863s]  INFO(runtime::session): no connection, starting idle kernel
[    15.990011s]  INFO(runtime::session): no idle kernel found

```

sbourdeauducq, 2018-09-23:
Is your JESD core up-to-date?

hartytp, 2018-09-23:
```
(artiq-dev) tph@buildsvr:~/scratch$ conda list
# packages in environment at /home/tph/anaconda3/envs/artiq-dev:
#
# Name                    Version                   Build  Channel
aiohttp                   3.1.3                    py35_0    conda-forge/label/main
alabaster                 0.7.11                     py_3    conda-forge/label/main
artiq                     4.0.dev0+1400.g69d060b6           <pip>
artiq-dev                 4.0.dev         1403+git92be9324    m-labs/label/dev
asn1crypto                0.24.0                   py35_3    conda-forge/label/main
async-timeout             2.0.1                    py35_0    conda-forge/label/main
asyncserial               0.1             py_13+git340e430    m-labs/label/main
attrs                     18.2.0                     py_0    conda-forge/label/main
babel                     2.6.0                      py_1    conda-forge/label/main
binutils-or1k-linux       2.30                          7    m-labs/label/main
blas                      1.0                         mkl  
bscan-spi-bitstreams      0.10.0                        2    m-labs/label/main
bzip2                     1.0.6                h470a237_2    conda-forge/label/main
ca-certificates           2018.8.24            ha4d7672_0    conda-forge/label/main
certifi                   2018.8.24             py35_1001    conda-forge/label/main
cffi                      1.11.5           py35h5e8e0c9_1    conda-forge/label/main
chardet                   3.0.4                    py35_3    conda-forge/label/main
colorama                  0.3.9                      py_1    conda-forge/label/main
coverage                  4.5.1            py35h470a237_1    conda-forge/label/main
cryptography              2.3.1            py35hdffb7b8_0    conda-forge/label/main
cryptography-vectors      2.3.1                    py35_0    conda-forge/label/main
dbus                      1.13.0               h3a4f0e9_0    conda-forge/label/main
docutils                  0.14                     py35_1    conda-forge/label/main
expat                     2.2.5                hfc679d8_2    conda-forge/label/main
fontconfig                2.13.1               h65d0f4c_0    conda-forge/label/main
freetype                  2.9.1                h6debe1e_4    conda-forge/label/main
gettext                   0.19.8.1             h5e8e0c9_1    conda-forge/label/main
glib                      2.55.0               h464dc38_2    conda-forge/label/main
gst-plugins-base          1.12.5               hde13a9d_0    conda-forge/label/main
gstreamer                 1.12.5               h61a6719_0    conda-forge/label/main
h5py                      2.8.0            py35hb794570_1    conda-forge/label/main
hdf5                      1.10.2               hc401514_2    conda-forge/label/main
icu                       58.2                 hfc679d8_0    conda-forge/label/main
idna                      2.7                      py35_2    conda-forge/label/main
idna_ssl                  1.0.0                         0    conda-forge/label/main
imagesize                 1.1.0                      py_0    conda-forge/label/main
intel-openmp              2019.0                      118  
jesd204b                  0.10                       py_1    m-labs/label/main
jinja2                    2.10                       py_1    conda-forge/label/main
jpeg                      9c                   h470a237_1    conda-forge/label/main
levenshtein               0.12.0                   py35_1    m-labs/label/main
libcurl                   7.61.0               h1ad7b7a_0  
libffi                    3.2.1                hfc679d8_5    conda-forge/label/main
libgcc-ng                 7.2.0                hdf63c60_3    conda-forge/label/main
libgfortran               3.0.0                         1    conda-forge/label/main
libgfortran-ng            7.2.0                hdf63c60_3    conda-forge/label/main
libgit2                   0.24.1                        7    m-labs/label/main
libiconv                  1.15                 h470a237_3    conda-forge/label/main
libpng                    1.6.35               ha92aebf_2    conda-forge/label/main
libssh2                   1.8.0                h5b517e9_2    conda-forge/label/main
libstdcxx-ng              7.2.0                hdf63c60_3    conda-forge/label/main
libusb                    1.0.20                        0    m-labs/label/main
libuuid                   2.32.1               h470a237_2    conda-forge/label/main
libxcb                    1.13                 h470a237_2    conda-forge/label/main
libxml2                   2.9.8                h422b904_5    conda-forge/label/main
lit                       0.4.1                      py_9    m-labs/label/main
llvm-or1k                 6.0.0                        25    m-labs/label/main
llvmlite-artiq            0.23.0.dev               py35_4    m-labs/label/main
markupsafe                1.0              py35h470a237_1    conda-forge/label/main
microscope                1.3                        py_1    m-labs/label/main
migen                     0.7             py35_73+gitbef9dea    m-labs/label/dev
misoc                     0.11            py35_31+git5ce139dd    m-labs/label/dev
mkl                       2019.0                      118  
mkl_fft                   1.0.6                    py35_0    conda-forge/label/main
mkl_random                1.0.1                    py35_0    conda-forge/label/main
[... 27 lines cut ...]
[... cut ...]
wheel                     0.31.1                py35_1001    conda-forge/label/main
xorg-libxau               1.0.8                h470a237_6    conda-forge/label/main
xorg-libxdmcp             1.1.2                h470a237_7    conda-forge/label/main
xz                        5.2.4                h470a237_1    conda-forge/label/main
yarl                      1.2.6            py35h470a237_0    conda-forge/label/main
zlib                      1.2.11               h470a237_3    conda-forge/label/main
```

hartytp, 2018-09-23:
(so, yes)

hartytp, 2018-09-23:
But, @marmeladapk's build that works predates the new JESD core, so I was wondering if the update was the issue...

sbourdeauducq, 2018-09-23:
Ok, then I have no better idea than building the version @marmeladapk,was using, checking if the problem is still there, and then bisecting

hartytp, 2018-09-23:
Okay, I'll do that. On the off-chance though, can you remind me where the best place to measure vccint is? I'd like to check that rework was done on the board I have, otherwise we'll be back into the bad old days of builds breaking with the phase of the moon

sbourdeauducq, 2018-09-23:
BTW, note that the new version of the JESD core is not compatible with the old - you need to have the matching ARTIQ version. The conda file in ARTIQ should tell you which JESD version to use.
VCCINT can be measured on the two large 0R resistors that are on the back side of the PCB and connected to the large coil.

hartytp, 2018-09-23:
0.96V so fine

sbourdeauducq, 2018-09-23:
You need to track the rustc version as well.

hartytp, 2018-09-23:
what's the easiest way of doing that?

Maybe it's easier to stick with the latest artiq and just manually revert individual commits.

hartytp, 2018-09-23:
Reverting those commits does indeed fix the issue.

sbourdeauducq, 2018-09-23:
Can you pinpoint which is causing the issue? I did test the new JESD core so this is surprising. Could it be vivado/P&R/timing problems?

hartytp, 2018-09-23:
> Can you pinpoint which is causing the issue

Yes, but it will take some time.

> I did test the new JESD core so this is surprising. Could it be vivado/P&R/timing problems?

My build scripts flag any timing issues as well as any vivado warnings that I haven't explicitly white listed. So far I haven't see anything suspicious.

hartytp, 2018-09-23:
Okay, building with everything reverted apart from the update to the new JESD core.

hartytp, 2018-09-23:
@sbourdeauducq what Sayma build did you test the new JESD core with? Standalone or master?

sbourdeauducq, 2018-09-23:
Standalone.

hartytp, 2018-09-23:
@sbourdeauducq remind me: are the commit strings in the artiq logs based on the actual git version it was built with or information from conda? ie do we trust that @marmeladapk's builds actually use the code revision in those logs?

sbourdeauducq, 2018-09-23:
I'm not aware of any bug with the version string.

hartytp, 2018-09-23:
ok

hartytp, 2018-09-23:
odd...still works with the new jesd core.

marmeladapk, 2018-09-23:
```
Commit ids:
artiq: 0e74194
misoc: 0a77fd4
migen: a6082d5
jesd204b: 03d3280
```

That's what I used + changed rx clock phase in misoc and commented out rtio sysref alignment in jesd204b.

hartytp, 2018-09-23:
> artiq: 0e74194

Your gateware is older than that according to the log I posted.

hartytp, 2018-09-23:
and, that uses the new version of the jesd204b core, which is compatible with the artiq revision you posted above, but not with the gateware revision reported in the logs I posted.

hartytp, 2018-09-23:
Okay, will have to finish this some other time.

I've started going through the list I posted above and reintroducing commits. So far I haven't been able to reproduce this issue.

hartytp, 2018-09-23:
I have no idea. Can't reproduce this any more. Will repoen the issue if I see it again, but for now this board seems to work.

marmeladapk, 2018-09-25:
I noticed the same error + failed prbs tests on MasterDAC variant.

hartytp, 2018-09-25:
Board damage? At least one board has these symptoms because of damaged coupling capacitors on the clock line (I'm beginning to think that these boards have become rather ragged from repeated rework and global shipping, and trying to get them to work might be a loosing battle -- time to spin some new ones up with the fixes properly implemented)

marmeladapk, 2018-09-25:
It worked just fine in Warsaw and here with the same gateware. Only thing that changed was gateware, I didn't even move it from crate.

marmeladapk, 2018-09-27:
It dissapeared. That's frustrating. I think I'm more conviced, that it's board fatigue.

hartytp, 2018-09-27:
@marmeladapk I looked at the output of the HMC830 on my board on a fast oscilloscope. It seems locked, but sometimes it jumps to a different frequency. I'll post a video in a second. But, if that is happening on other boards it could explain some of the issues we've seen!

NB I'm running my HMC830 output at 2.4GHz not 1.2GHz

hartytp, 2018-09-27:
Probing on the UFL connectors.

It was at 2.4GHz at first, but then it jumped and is now wandering between 2.9GHz and 3GHz.

https://drive.google.com/open?id=1OgUiDcEbpp-GNcApitAWPpE_TnNefFv1

The signal at the reference input looks a bit weaker than I expected (only about 400mVpp)

hartytp, 2018-09-27:
Okay, another observation: the UART reports that the HMC830 is locked but when I trigger a scope from the HMC830 reference input and look at the RF output on J59 it's pretty clear that it isn't locking even transiently.

I don't have a fast enough probe to look at the HMC830 output directly so it's possible that the 3GHz RF I'm seeing is actually just a clock buffer free-oscillating.

hartytp, 2018-09-27:
@marmeladapk is there any way you can track down the board I sent you and post it to me? That board didn't have any of these issues, and this is just killing me

gkasprow, 2018-09-27:
@hartytp I observed output with spectrum analyser and to make sure it locks, I changed the input frequency by a few tens of Hz to observe if output follows it

hartytp, 2018-09-27:
@gkasprow 
- this may actually be a red-herring since the signal I see is there even without the RF source present. So it's the HMC830 or one of the clock chips free-oscillating. Interestingly, it doesn't start oscillating until around the point in the boot up where the serwb connection starts
- I agree that I've definitely seen the HMC830 locked before. I've even measured its phase noise spectrum, so I know it can lock. What I don't understand is why it won't lock on this board.

hartytp, 2018-09-27:
Cool. Poked around a bit to find what was oscillating and it started working! Whoo!

hartytp, 2018-09-27:
Anyway, @gkasprow one thing I still find concerning is that it is clearly possible for the HMC830 to indicate lock when it's not in fact locked. That's a pain and important to know

closed 2018-11-27 by hartytp