sinara-hw/sinara #586: hmc830 lockups

hartytp, 2018-09-21:
I haven't figured out the exact sequence of events, but here is a rough observation about the HMC830:
- it is possible to make it go into lockups where it does not lock. It identifies correctly and the register dumps look fine, indicating that it is in the correct SPI mode
- I do not know what causes this
- power cycling the board briefly does not fix it
- leaving it unplugged for a while with the RF source also off seems to fix it. I do not know if switching the RF off is really required, but it was a case of turn everything off, get a coffee and come back to a working board

This is one more thing that makes developing on Sayma a PITA, so it would be great to fix it for the next revision.

sbourdeauducq, 2018-09-22:
Is there really no alternative to this chip? HMC silicon seems to be very crappy.

marmeladapk, 2018-09-22:
@hartytp How many AFEs do you have plugged in? In noticed that it was more common with 4 AFEs than with 2.

hartytp, 2018-09-22:
> Is there really no alternative to this chip? HMC silicon seems to be very crappy.

There are plenty of alternatives. Just a question of whether we really want to invest the time in replacing the chip, writing code for the replacement and then doing careful testing. And, even then, we could find that the new chip has its own "features" which cause similar problems.

hartytp, 2018-09-22:
> @hartytp How many AFEs do you have plugged in? In noticed that it was more common with 4 AFEs than with 2.

0

But, that's quite concerning, isn't it. I hadn't heard about this before, but maybe it points to some PI issue. This sounds like something we need to fix before v2.0.

hartytp, 2018-09-27:
I have a feeling that there may be a correlation between this issue and applying RF before DC power. Small statistics, so maybe it's just coincidence.

hartytp, 2018-09-27:
If anyone can reproduce that it would be a great help!

gkasprow, 2018-09-27:
0.5s is far too little time. Capacitors won't discharge and the chip may enter some latch up state.

hartytp, 2018-09-27:
@gkasprow in hindsight, that's obviously true. I should have been more careful. However, this does point out how much of a pain the HMC830 is: it's the only chip we have that has state that cannot be reset in any way other than a power cycle, which is ridiculous.

I think we either need to choose a different PLL or have some means of power cycling it after the FPGA has successfully booted.

gkasprow, 2018-09-27:
The 5V rail has low power consumption. So the voltage can be present long time after the power is off.

hartytp, 2018-09-27:
hmm...on this board, I often see the HMC identification error after flashing with `artiq_flash` (no power cycling). I never saw that before.

hartytp, 2018-09-27:
Anyway, the answer is basically that it's pretty easy to put the HMC830 into a locked up sate that it can't recover from.

gkasprow, 2018-09-27:
There should be 10k resistors attached to one of HMC chips SPI interface lines that ensure that no SPI configuration is selected by chance.

hartytp, 2018-09-27:
looks like that may have come loose in shipping. I'll touch it up and see if the issue goes away. Thanks for the pointer!

jbqubit, 2018-10-04:
I have not seen this type of HMC830 lockup on the three boards that span Duke and UMD. Will keep an eye out for it.

hartytp, 2018-10-04:
AFAICT the issue is that the HMC830 is quite fussy and easy to brick: it doesn't like brown-outs or odd power sequencing; it doesn't like any garbage being written to its serial port or serial transactions being interrupted; etc Any of these conditions put it into a state that can only be recovered by power cycling.

If we decide to keep it for the next hw rev then I think we should add some fets to allow it to be power-cycled after the FPGA has booted, as previously discussed.

dhslichter, 2018-10-04:
@hartytp do we anticipate that a different chip, e.g. ADF4351, one would not see these issues?  Can one see them well enough with an HMC830 eval board to be able to believe that a test with an ADF4351 eval board would turn up equivalent-type problems it might have?

hartytp, 2018-10-04:
> @hartytp do we anticipate that a different chip, e.g. ADF4351, one would not see these issues?

Not these issues, but doubtless other ones. The downside of the ADF4351 is that it's a much more complex IC and, as we've seen over and over again, that can lead to issues.

>  Can one see them well enough with an HMC830 eval board to be able to believe that a test with an ADF4351 eval board would turn up equivalent-type problems it might have?

Nope. The hMC830 eval board behaved fine when we tested it. The misbehaviour in Sayma is more to do with corner cases like supply sequencing, etc which are hard to test on an eval board. 

AFIAK Mirny is likely to be funded soon (already funded) and will have this PLL, so that will be a better test case for using this PLL with ARTIQ -- that's one of the main reasons I suggested it as a possible replacement.

Note that I'm not arguing that we necessarily should switch the PLL, just documenting the issues I've seen and raising the issue for discussion before Sayma v2.0. My guess is that adding some FETs to allow the HMC830 power supplies to be held low until the FPGA has booted would resolve all these lockup issues.

As I noted elsewhere, a potentially more important issue about the HMC830 is that its output divider is not in the feedback path. So, if one wants to clock the DAC below 1.5GHz using the divider the DAC clock phase won't be deterministic. So, we need to do one of: (1) not use DAC clocks below 1.5GHz (always use interpolation); (3) use a different PLL with the divider in the feedback path; (2) use the RTM FPGA to sample the DAC clock <-> reference clock phase shift and reset the divider (toggle between two different values) until it has the correct value. Probably the correct option is to go for (1) in the first instance, but make sure the hardware is set up for (2) in case we need that later. But I'm open to changing the PLL if someone makes a strong case for it.

dhslichter, 2018-10-05:
Since I am not in the trenches on this one I won't try to push for a particular solution, but if we really believe that power supply sequencing will address all of the HMC830 bugs, then it seems like we are better off sticking with it.  I also worry about the time to get the ADF4351 up and running if we used it to replace the HMC830, as Sayma has already had enough delays.  But again, people working with the chips directly are in a much better position to comment on how accurate these sentiments are.

hartytp, 2018-10-05:
>  Look, the HMC830 garbage has wasted more time than the much more complex

@sbourdeauducq Agreed that, in hindsight, the HMC830 was the wrong chip to use and has caused us all a huge headache.

However, the relevant question now is not "how much time has it wasted", but rather "how much more time will it waste?" or even "is the expected wasted time in the future lower with the HMC830 or with some other PLL IC that we have no experience with yet"

My belief is that we now understand pretty well how to deal with the HMC830 bugs: keep it powered down until the FPGA has booted and then always program all registers at startup. That's a bit of a pain as it will requite a bunch of FETs to sort out the power sequencing and then probably also require us to route a PG line to the FPGA (or, I guess, measure the time taken for the power rails to settle and then hard-code a delay in the firmware). But, it does seem likely that once that's done we will no longer have any issues with the HMC830.

In contrast, another PLL will require a fair amount of design work, more firmware to be written, more testing, etc. And it would be easy to mess it up and cause problems.

I do not want to force us to use the DAC interpolation -- I'm still concerned that we could see an AD9914-type bug where the DAC craps out at max clock speed -- , so we do need to ensure that we can synchronize the output divider. That's more work, but it's something we've done before and know how to do, so it's not high risk.

sbourdeauducq, 2018-10-05:
Would it be difficult to have both on the PCB and then select one or the other e.g. by changing a few capacitors etc.?
Then we can have a HMC830 demo with the existing code, while we work on bringing up the ADF4351, and if the latter turns out to be another trash fire we can keep the HMC.

hartytp, 2018-10-05:
> Would it be difficult to have both on the PCB and then select one or the other e.g. by changing a few capacitors etc.?

It could be done, although it will make the layout more of a mess.

> Then we can have a HMC830 demo with the existing code, while we work on bringing up the ADF4351,

I guess you would want more funding to work on the ADF4351? I'm guessing that adding some simple code to reset the HMC830 divider would be much cheaper to fund that setting up the ADF4351 from scratch and, currently, seems like the lower risk path

sbourdeauducq, 2018-10-05:
> I guess you would want more funding to work on the ADF4351?

Considering that most competent developers we could hire demand at least 5000 EUR/month (and more usually 8000), doing otherwise does not sound like a reasonable decision, does it?

hartytp, 2018-10-05:
Right, it would definitely need funding. What I'm getting at is that it's likely to be relatively expensive to fund ADF4351 development. It's reasonable to think that people aren't going to want to fund this kind of development unless a strong case can be made for it. And right now, I don't think such a case has been made: if all remaining HMC830 issues are fixed by adding a power cycler -- and I'm relatively confident that they will be -- then what advantage does it offer? What's a higher risk: that the HMC830 will continue to play up after adding a power cycler, or that the ADF4351 will have some as yet unknown issues that will waste time? Knowing what we know now, I'd say the latter, wouldn't you?

sbourdeauducq, 2018-10-05:
Maybe. But considering that the HMC830 is a piece of garbage and needs hacks that are a bit ugly, and that the alternative chip will be most likely supported for Mirny already, I like to keep the option of ditching HMC open.

dhslichter, 2018-10-05:
How about the following: we go with the HMC830 + hacks for Sayma 2.0, and in parallel the ADF4351 is developed for Mirny.  If we discover that for some reason our hacks are not sufficient, or that the ADF4351 is a much better choice (after probing for bugs) than HMC830 + hacks, we can change it out on the next generation of Sayma.  I think it's probably challenging to put pads on for the ADF4351 in this version, given that we know nothing about it and what tweaks it might need, and have that end up working correctly.  

And yes, Tom, thank you for pointing out the sunk cost fallacy.  It's absolutely infuriating and maddening what has happened with the HMC830, and of course in hindsight we would never use it, but given where we are, the question has to be what is the simplest/highest percentage shot at this moment in time.  

Thoughts?

hartytp, 2018-10-05:
What about this: we keep the HMC830 + power cycling FETs as plan A. Then we add enough test points for power + signal + ground + SPI that we could easily glue on a small PCB with the other PLL on and hack it in.

dhslichter, 2018-10-05:
@hartytp what about SI for the clock signal?  Are you going to put some sort of proper connector for the clock signal from such a "hacked on" board?  I would think it'd be important, and then this is more of a PITA.  Probably a SATA connector or similar would work, enough pins for power/ground/SPI and reasonable impedance control for the clock signal?  0402 solder jumpers on the RTM to choose whether clock comes from HMC830 or from connector.

gkasprow, 2018-10-05:
We can keep the idea of clock mezzanine...

gkasprow, 2018-10-05:
We can also place SMT pads around and install fix PCB as castellated module if we are not happy with MHC830.

hartytp, 2018-10-05:
> We can keep the idea of clock mezzanine...
> We can also place SMT pads around and install fix PCB as castellated module if we are not happy with HMC830.

That seems like a lot of extra work, complexity and expense which isn't justified in this case: a whole mezzanine board in case we can't make a standard PLL work properly?!

> @hartytp what about SI for the clock signal? Are you going to put some sort of proper connector for the clock signal from such a "hacked on" board? I would think it'd be important, and then this is [...]

@dhslichter that was only a suggestion as an absolute last resort for the next prototyping round in (what I believe is an extremely unlikely event) that the fixes we add don't stop the HMC830 causing us further trouble.

My point is that if you really need to you can hack these things in without too much grief (even if you solder some short bits of coax between boards). But, again, I don't think it's at all likely that we will need to do this, so no point adding extra connectors and routing for it.

hartytp, 2018-10-05:
@sbourdeauducq I have a huge amount of sympathy for your point of view here. Having lost days to that thing crapping out in annoying ways I've started to wish unspeakable ills on whoever designed its digital interface (the analog side is ace though).

But, my concern here is that desire to take revenge on that IC is the main driver for removing it, rather than practical considerations.

Having spent some time working with it, I believe the lock ups are due to:
- improper power cycling (e.g. turning the power supply back on before all rails have discharged). Adding some FETs to ensure the power rails are pulled low until they are enabled by the FPGA will fix this
- glitches on the CS/CLK lines. Fixed by holding it in a powered down state until the FPGA has started
- writing junk to the SPI or aborting SPI transactions early (I did this a few times when setting up the delay line SPI). Easiest is to put it on a separate SPI bus in gateware. Again, the power cycler makes sure that even if something nasty does happen it won't cause failures on subsequent boots

Adding a few FETs is a little ugly, but basically fine IMHO.

The bigger issue for me with the HMC830 is the unsynchronised dividers. But, correct me if I'm wrong about this, we are pretty confident we can use the FPGA to detect the phase and synchronise the dividers up to 1.5GHz (the lower bound of the VCO range). The hard part of that has been implemented in gateware/firmware already for the SiPhaser and the HMC7043 sayma sync, so while it's still more work to implement, it's not too bad and there aren't any scary unknowns.

In contrast when I actually think about the work involved in switching to a new PLL my heart sinks a little. Someone's got to read the datasheet carefully, think it all through, do the design inc PLL simulations, check against the eval boards since data sheets are usually ambiguous and incomplete. Then it has to be incorporated and tested and we have to hope that nothing nasty crops us. True, it's nice to share that with Mirny but, AFAIK Mirny isn't actually funded yet. And, even when it is, we won't have the actual HW to test before Sayma v2.0 goes off to manufacture, so we won't find out about any bugs until it's too late to fix them in the next hw revision.

Right now, I'd prefer to focus our energies on the really crucial things like the ramp generator glitches and synchronisation issues. What do you think?

sbourdeauducq, 2018-10-06:
> The hard part of that has been implemented in gateware/firmware already for the SiPhaser and the HMC7043 sayma sync, so while it's still more work to implement, it's not too bad and there aren't any scary unknowns.

Er, no, you cannot extrapolate like that. One of the hard parts will be to make the HMC830 change its phase without breaking. How many resets are needed? Are they fast enough? And Ultrascale and its huge PVT + place-and-route variation may mean boards may need frequent and annoying recalibration. This is another ugly hack on top of all the other ones.

On Urukul we already did clock rerouting by changing capacitors. Can't we do the same here and put both the HMC830 and ADF4351 on the prototype PCBs, then DNP the worst one on production boards?

hartytp, 2018-10-06:
@sbourdeauducq 

> On Urukul we already did clock rerouting by changing capacitors. Can't we do the same here and put both the HMC830 and ADF4351 on the prototype PCBs, then DNP the worst one on production boards?

You're right, I probably am overestimating how hard it would be to have two PLL chips and use solder bridges/capacitor jumpers to select between them. Let's say for now that we do that.

What worries me more is who will do the analog design. The ADF4351 was a chip I picked by sorting PLLs by phase noise and picking the first one where the block diagram had dividers in roughly the right place, the fact that it's also planned to be on Mirny was the icing on the cake. However, I have not read the data sheet in detail or given it too much real thought. Before we add it to Sayma, someone needs to check the available parts and make sure this really is the appropriate choice; read the data sheet in detail and figure out how we would connect it, what modes we would use, make sure it will actually do what we want; check how the eval board is connected to make sure we haven't missed anything (c.f. Zotino/the HMC7043's "debug modes" where it's easy to miss some subtlety in the data sheet and produce a design that doesn't work); simulate loop filters; commit to testing it when it arrives; etc. Who will do that work?

It's all very doable, but easy to underestimate how time consuming it is and how easy it is to miss some small detail which results in a broken design. Will somebody provide additional funding for this work?

hartytp, 2018-10-06:
> Er, no, you cannot extrapolate like that. One of the hard parts will be to make the HMC830 change its phase without breaking. How many resets are needed? Are they fast enough?

I don't think that's so hard. It's just firmware and easy to test. I'm happy to write a HMC830 phase reset script and test it this week. I would do the following:
- VCO at 2.4GHz
- output divider at 4x to give 600MHz f_out
- add a phase reset function which sets the divider to 1x and then back to 4x
- put a loop in the init function that resets the phase once per second
- observe the output on a fast scope triggered from the 150MHz clock
- verify that the phase does jump randomly between the 4 possible values with roughly equal probability. Hence any phase can be achieved in of order 8 times the duration of an SPI write. NB we don't have to wait for the PLL to relock

If this works reliably over several power cycles, I think we can conclude it's not a problem. I don't anticipate issues, but you're right to be cautious; we should check.

> And Ultrascale and its huge PVT + place-and-route variation may mean boards may need frequent and annoying recalibration.

Wouldn't we do this on the RTM FPGA? That's where the SYSREF generator will be, so it makes sense to do all the clocking on that once DRTIO is implemented.

> This is another ugly hack on top of all the other ones.

Life is an ugly hack.

> I'm not even sure how this fits into the global clocking architecture - what do you compare the HMC830 output with? Additionally, high frequencies will require using the IOSERDES which is yet another can of worms (especially on Ultrascale).

Well, that's fair. The question from a funding/planning point is: which path is lower risk/less work/requires less funding/will be quicker to make reliable?

I suspect that all of our judgement is biassed towards overestimating the difficulty involved in the parts which would be our responsibility. But, if you really feel that the different PLL would be less work/less time, and if we can find some funding for someone to implement the new PLL then it's fine by me.

sbourdeauducq, 2018-10-06:
The phase check in gateware needs funding as well. The hardware work for the ADF4351 is required for Mirny already.

hartytp, 2018-10-06:
> Please don't have this approach. This path leads to designs such as HMC chips, Xilinx transceivers, $10k RFSoCs that get physically damaged if you program their CSRs in the wrong order, and user interfaces like this:

Eye wateringly nasty.

> The phase check in gateware needs funding as well.

ACK. Rough cost estimate for that?

> The hardware work for the ADF4351 is required for Mirny already.

Is that actually funded yet (@jordens)? Who is taking responsibility for doing design reviews on Mirny? Will the design for Mirny be ready before we need to start freezing the design for Sayma v2.0 (next couple of weeks hopefully). Who will take responsibility for assessing what design changes are required to adapt the Mirny design to Sayma, which will have somewhat different requirements? Once there is a core device driver, how much additional funding will be required to adapt the Sayma firmware to use the ADF4351?

jordens, 2018-10-06:
So many questions. AFAIK there is currently only general interest in Mirny.

hartytp, 2018-10-07:
Well, @sbourdeauducq, I'm not sure that there is much more I can add to this discussion.

tl;dr we either keep the HMC830, add the power cycler FETs and gateware to reset the output divider, or we swap the HMC830 for a new PLL chip. Both paths require some level of funding. The latter seems higher risk and somewhat more expensive to me, but I may well be underestimating the complexity in synchronising the HMC830 phase.

If you have a strong opinion about this, I'm happy to go with it, so long as we can secure sufficient funding (@jbqubit) to make sure we do a proper job of it.

sbourdeauducq, 2018-10-07:
My opinion is that we should have both PLLs on the PCB (switched with replaceable capacitors) to maximize our chances of ending up with a well-working board. But yes, that requires funding.

hartytp, 2018-10-07:
Sure. Some obvious points, but worth being clear:
1. Putting the new pll on as an option requires as much work from a hardware design point of view as putting it on as the only option. No point including it if we haven't done a careful design
2. We have to decide what our plan 'a' for software support will be (old pll with fpga sync or new pll) and ensure it's correctly funded before committing to anything.

Basically, let's have a clear plan and confirm funding before committing to anything.

sbourdeauducq, 2018-10-07:
Plan A is another PLL. HMC and dirty hacks need to die.