sinara-hw/sinara #588: sayma: sysref coupling + termination

hartytp, 2018-09-27:
Sysref should be DC coupled and 50Ohm terminated at the DAC. Currently, it's AC coupled and untermianted (AFAICT, the DAC sysref inputs are high-Z).

hartytp, 2018-09-29:
NB need to be careful with the logic level conversion here since the optimal common-mode voltage for the SYSREF inputs is 600mV. So, need a resistor network to match that to the LVPECL/CML/whatever driver.

NB in my tests, I'm using AC-coupled LVPECL since it would be very tough to add such a network to Sayma (too fine pitch everywhere, no convenient access to ground, etc).

gkasprow, 2018-10-08:
It's easier to remove terminating resistors and switch to LVDS mode. At the moment the termination resistors are soldered directly to the DAC pins. AFAIR I did it not only for clock but also for sysref inputs.

marmeladapk, 2019-01-09:
@gkasprow @hartytp What is the conclusion here? In schematics SYSREF is AC coupled, has 100R differential termination and 200R to ground at source.

Source:
[image]
DAC:
[image]

hartytp, 2019-01-09:
Closing in favour of https://github.com/sinara-hw/Sayma_RTM/issues/13

closed 2019-01-09 by hartytp

gkasprow, 2019-01-09:
200R is pulldown for open emitter outputs. 100R is differential termination. Such a solution consumes far less power than Thevenin termination.