sinara-hw/sinara #589: silk screen

hartytp, 2018-09-27:
Please can we make sure that all the important components are labeled on the silk. I'm trying to trace an issue with the clock network on my Sayma, but it's taking me ages to figure out which mux is which.

hartytp, 2018-09-28:
@gkasprow out of curiosity, why are all the decoupling caps on Sayma 0201? In a lot of places we could go to 0402 without issue (and make life much more livable for anyone trying to do rework).

gkasprow, 2018-09-28:
0402 make worse SI - the bigger the cap, the greater the impedance discontinuity.
And since 100nF 0201 were used on gigabit lines, there was not much sense to add another BOM line for 0402 version of them

hartytp, 2018-09-30:
@gkasprow I can't comment on the 10GBPS links, so if you tell me that 0402 makes a noticeable difference compared with 0201 then I'll believe you (particularly since I know how much time you spent simulating SI in those lines).

However, for the DAC SYSREF pins, there is obviously no SI issue with 0402 capacitors. For that matter, 0805 would be fine there -- I'm using 0805 capacitors on stripboard without issue at the moment.

For standard RC components like 100nF capacitors, is there a noticeable cost associated with adding lines to the BOM? i.e. will there really be a noticeable cost increase associated with, say, using both 0603 and 0201 capacitors?

The advantage of 0603 is that it makes rework much easier for those of us who don't have high-end rework stations. It's not unlikely that we'll need to make some small fixes on Sayma v2.0 to correct for issues we haven't discovered yet and, based on my experience with Sayma v1.0, increasing via sizes and component sizes a little would make a big difference to how easy that kind of work is.

sbourdeauducq, 2018-09-30:
I don't find 0201 extremely hard to solder with just a regular iron and a fine tip, especially if the pads are a bit on the large side (I have not done it on Sayma, maybe a good compromise is to use larger pads?). One advantage of 0603 and larger sizes is that resistors are marked.

gkasprow, 2018-09-30:
I don't use soldering iron for most of the work.
Good temperature controlled hot air gun (30EUR) + precise tweezers + microscope + soldering flux do the job nicely and do not leave any traces after rework. In such case it does not mater if you use 0201 or 0805.

gkasprow, 2018-09-30:
Another thing about 0201 - it's hard to fit 0402 between vias under BGA.
I can replace some caps with 0402, but most decoupling ones below BGAs want to leave 0201.

hartytp, 2018-10-02:
@sbourdeauducq 0201 is fine to solder if it's just one component. But, when you have a development board that needs rework it can become  a pain. e.g. for the synchronisation I needed to solder some coax onto 0201 pads and then construct a RC network to terminate and match the signal to the correct logic levels for the SYSREF input. There just physically isn't much space around those pads to do that and without proper optics it's hard to see what you're doing. Moreover, the pads can delaminate easily after repeated rework and don't offer much mechanical strength for attaching things to. 

Anyway, if people really want to keep the 0201 components it's okay. My point is just that Sayma has required quite a lot of rework and that would have been easier with larger components slightly wider separated.

jordens, 2018-10-02:
Looking at the trade off between designing for easy hacking (add 0402 BOM line, change schematics, change layout, iterate and debug, discuss issues) and streamlining it for manufacturing (status quo), I'd really go for the latter. If Sayma 2.0 needs to be easy to hack beyond the usual use, then we should specifically design a prototype for hacking and debugging.

hartytp, 2018-10-02:
@jorders while I take your point, I think it's important not to underestimate how much time could have been saved in debugging Sayma v1.0 if we'd had a few niceties like test points etc.

Even if we don't change any resistor pads, there are a few things that can be done quickly and without the need for in-depth discussion, which aid debugging without affecting manufacturability:
- label all important IC2 on the silk
- add test points to all power rails, slow busses (I2C/SPI) and other important signals. where possible also aim to expose all clocks via ufl or similar (e.g. if there is an unused output on a clock buffer then connect it to a UFL test point)
- for slow signals, don't make traces + vias unneccesarily small and don't cram components too close together (obviously, feel free to ignore this where it makes a genuine difference to SI/ease of layout, but don't just cram things together where they don't need to be)
- where possible (particularly slow signals) add 0R resistors that can be used to disconnect the signal and allow injection of an external signal. also consider adding 0R/DNP pull-ups/pull-downs on logic lines

etc. Just standard good practice. This kind of thing makes debugging the EEMs much easier than debugging Sayma

jordens, 2018-10-02:
Absolutely. I am all for test points, labels, and general testability layout. IME the cost/benefit threshold for adding such a test/debugging item is reached roughly when the item will likely be useful (look up refdes, place/remove 0R, probe a voltage, attach some hack, find a labelled connector...) at least twice.
But it really pays of to do it from the start, i.e. in the schematic. Otherwise it's a pain.

hartytp, 2019-03-12:
@gkasprow is this done? Can we close? Or should we move to the new repos?

gkasprow, 2019-03-13:
I

closed 2019-03-13 by gkasprow