sinara-hw/sinara #592: sayma: heatsinks

hartytp, 2018-10-01:
With the DACs running at 2.4GHz they get scorching hot, even with decent air flow. Might be worth small heat sinks in the next version (also for the RTM FPGA?).

dhslichter, 2018-10-02:
Sounds like a good idea -- @gkasprow is there enough space for low-profile heat sinks to be mounted properly?

gkasprow, 2018-10-02:
The chips dissipate the heat mostly via bottom side thermal pad. I'm not sure if heat sink on top would give any result. We can place copper thermal pad on bottom and there is a lot of space for small heatsink. On top side there is also some place but we would need to make sure that we don't collide with AFE components. What can be improved easily  is number of thermal vias below DAC.
 Doesn't the DAC chip have the temperature sensor? I'm curious what is its temperature. Take it into account that 55 degrees is hot for human but harmless for the chip.

hartytp, 2018-10-02:
> Take it into account that 55 degrees is hot for human but harmless for the chip.

I'm not worried about damage to the chip. I'm more worried about the amplitude and phase stability of the RF, which needs to be very good for quantum experiments. Elevated temperatures usually lead to worse thermal stability.

> What can be improved easily is number of thermal vias below DAC.

:+1:

> We can place copper thermal pad on bottom and there is a lot of space for small heatsink

Let's at least leave a pad for a heat sink. We can always DNP it if we decide it's not necessary.

gkasprow, 2018-10-02:
let's leave this issue open.

dhslichter, 2018-10-03:
> I'm more worried about the amplitude and phase stability of the RF, which needs to be very good for quantum experiments. 

Agreed, this is not a chip damage issue, it's an output stability issue.  Putting a good copper thermal pad on the bottom sounds like a reasonable solution.  Also, measuring the DAC temperature with the onboard sensor sounds like a good plan.  The absolute temperature really doesn't matter that much, as long as it is in the specified operating range for the chip; the main concern is temperature stability.  DAC gain coefficient is -114 ppm/C, which is relatively large as these things go -- 8 LSB per degree C.

gkasprow, 2018-11-28:
Once we remove ADCs, we can mount flat heat spreaders that will improve heat transfer between DAC and air. I will reserve space for them. They can be soldered to the thermal vias under DACs.

hartytp, 2019-03-12:
@gkasprow is this done? Can we close or move to the new repo?