sinara-hw/sinara #593: sayma: replace HMC830

hartytp, 2018-10-02:
What about replacing the HMC830 with a different PLL like the ADF4371 or simillar? The advantages I can see are:
- this is the PLL that will be used in Mirny, so if we also use it on Sayma then we can share hardware/software development between the two projects
- afaict (someone needs to check this) the output divider can be placed inside the feedback path, so the latency will be deterministic even when the output frequency is a division of the VCO frequency. That means that we don't have to worry about resetting the PLL to achieve deterministic latency between the DAC clock and the reference clock
- it *may* have a less crappy interface than the HMC830 that's less prone to lock ups

Downsides:
- it's an unknown so should cause delays. But, if Mirny gets funded then that debugging doesn't necessarily have to be funded from Sayma's budget
- more work needed on Sayma v2.0. But, the hardware work required to replace it will be much less than the time we've spent debugging obscure issues with the HMC830

hartytp, 2018-10-02:
- HMC830 phase noise for 3GHz/30=100MHz: -142dBc/Hz at 100kHz offset; -165dBc/Hz at 1MHz offset
- ADF4371 phase noise for 4GHz renormalized to 100MHz: -149dBc/Hz at 100kHz offset; -172dBc/Hz

So, the VCO open-loop noise for the ADF4371 looks fine. Need to check the contributions from the PFD etc...

hartytp, 2018-10-02:
Note that this is a more complex PLL than the HMC830, so our motivation for using it would largely be that we can recycle work that's already being done for Mirny, which then becomes a kind of eval board for this.

hartytp, 2018-11-27:
@gkasprow Before we start implementing this on the hardware, I need to double check that the ADF4371 is definitely the PLL we want to use as an alternative to the HMC830...

gkasprow, 2018-11-27:
We cal always leave clock mezzanine and make adf chip board

sbourdeauducq, 2018-11-27:
> I need to double check that the ADF4371 is definitely the PLL we want

If not, we should definitely look for a non-HMC alternative...

hartytp, 2018-11-27:
> If not, we should definitely look for a non-HMC alternative...

Yes. FWIW, I think the ADF4371 is the one we want, I just haven't done sufficient due diligence yet.

hartytp, 2018-11-27:
> We cal always leave clock mezzanine and make adf chip board

As discussed before, I'd really prefer not to do that. Putting the PLL on a mezzanine has a lot of overhead in terms of connectors etc and just isn't worth it. Let's just use solder jumpers to select between the HMC830 and the ADF...

gkasprow, 2018-11-27:
Btw, the  ADF4371 is not available yet. And as a new product it may have a lot of features :)

hartytp, 2018-11-27:
@gkasprow ack. As I said, give me a little time to survey the available options. I'll post back with some options in the next few days.

marmeladapk, 2018-12-13:
@hartytp AFAIK this is duplicate of #602 with no additional info over #602. Closing.

closed 2018-12-13 by marmeladapk