sinara-hw/sinara #595: Meta: Sayma v2.0

hartytp, 2018-10-02:
Okay, we now have reliable RF outputs from Sayma and a demo of board to board synchronisation. It's time to start getting ready for Sayma v2.0!

What other tests do we need to do on the current Sayma boards/what known problems do we need to fix? The ones I'm aware of are:
- [ram issue on some boards](https://github.com/m-labs/artiq/issues/1124)
- [si + mechanics for new AFE form-factor + connectors](https://github.com/sinara-hw/sinara/issues/386)
- [Sayma power in utca racks](https://github.com/sinara-hw/sinara/issues/571) fixed?
- [issue with powering VHDCI breakout cards](https://github.com/sinara-hw/sinara/issues/587)

So, not too much to do before we can start work on updating the design...

gkasprow, 2018-10-02:
@hartytp the test boards for AFE v2 are already assembled. I will get them this week. Once I check that they work, will post you the set.

hartytp, 2018-11-27:
@jbqubit any updates on the uTCA power issues?

hartytp, 2018-11-28:
Closing this issue, since there is nothing here apart from https://github.com/sinara-hw/sinara/issues/386 which is currently waiting for @jbqubit to post a video of Sayma booting in his rack.

closed 2018-11-28 by hartytp