sinara-hw/sinara #596: Urukul (AD9912) glitches during phase ramping

kjh-m, 2018-10-11:
Hello,

I'm getting glitches from AD9912 output during a phase ramp experiment. Phase is ramped by small steps at a given intervals, while keeping the frequency set-point constant.

Problem persist for wide range of interval times (10 us - 10 ms) and frequencies (100 kHz to 300 MHz).

Initially I noticed this while doing frequency measurement with a counter. Observing the waveform with an oscilloscope revealed fast glitches occurring at random points of the cycle (see below).

[image]
[image]
[image]

Zooming in:
[image]

It seems that the glitches don't occur at every phase step. Screen capture of the oscilloscope shows glitch interval of 728 us, which is weirdly reproducible. In the measurement below, scope is triggered from glitches, so the waveform itself actually runs freely.

[image]

Also, DDS doesn't seem to lose coherence at the glitch, and the waveform continues smoothly.

All electrical connections are terminated to 50 ohms. Urukul is referenced to an external clock, while Kasli is not.

Simplified version of the phase ramping program is shown below.

- Kalle

----

```
from artiq.experiment import *

class stepper(EnvExperiment):

    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_cpld")
        self.setattr_device("urukul0_ch0")

    @kernel
    def run(self):

        self.core.reset()

        self.urukul0_cpld.init()
        self.urukul0_ch0.init()

        delay(1*ms)

        self.urukul0_ch0.set_att(10.)

        freq = 1*MHz
        freq_word = self.urukul0_ch0.frequency_to_ftw(freq)

        # Step phase by 16 units every 1 ms
        step_time = 1*ms
        phi_step = 16
        phi = 0

        # Start DDS
        self.urukul0_ch0.set_mu(freq_word,phi)
        self.urukul0_ch0.sw.on()

        delay(step_time)

        while True:
            # Increment phase
            phi += phi_step
            # Check if phase wraps around 2*pi, and correct accordingly
	    # 2*pi equals 2^14 = 16384
            if phi >= 16384:
                phi -= 16384
            # Set phase word
            self.urukul0_ch0.set_mu(freq_word,phi)
            delay(step_time)
````
----

jordens, 2018-10-11:
This issue was moved to sinara-hw/Urukul#15

jordens, 2018-10-11:
@gkasprow could you close this or give me permissions?