sinara-hw/sinara #597: Flashing MMC using JTAG

jbqubit, 2018-10-12:
Trying to flash MMC firmware from [here](https://github.com/sinara-hw/sinara/issues/571#issuecomment-426252827). Using [these](https://github.com/sinara-hw/meta/wiki/Sayma#flashing-mmc) instructions. With MMC JTAG connected supply current draw is only 24 mA.

I see NXP programmer...

```
$ dmesg
[1280676.704387] cdc_acm 1-9:1.1: ttyACM0: USB ACM device
[1280676.704540] usbcore: registered new interface driver cdc_acm
[1280676.704541] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[1281581.096725] usb 1-9: USB disconnect, device number 43
[1281582.607374] usb 1-9: new high-speed USB device number 44 using xhci_hcd
[1281582.756368] usb 1-9: New USB device found, idVendor=1fc9, idProduct=000c
[1281582.756371] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1281582.756372] usb 1-9: Product: LPC
[1281582.756374] usb 1-9: Manufacturer: NXP
[1281582.756375] usb 1-9: SerialNumber: ABCD
```
LPCxpresso flash tool reports "1 emulator found" LPC-LINK2 CMSIS-DAP V5.173. However, programming doesn't work. 

```
Ni: LPCXpresso RedlinkMulti Driver v8.2 (Sep  5 2016 10:13:28 - crt_emu_cm_redlink build 725)
Pc: (  0) Reading remote configuration
Pc: (  5) Remote configuration complete
Nc: connecting to core 0 with handle 0x0 (using server PID -1) gave rc 'Ee(36). Could not connect to core.'
Nc: connecting to core 0 with handle 0x0 (using server PID -1) gave rc 'Ee(36). Could not connect to core.'
Nc: connecting to core 0 with handle 0x0 (using server PID -1) gave rc 'Ee(36). Could not connect to core.'
Nc: probe unbooted? no core 0?
Ed:02: Failed on connect: Ee(36). Could not connect to core.
Et:31: No connection to emulator device
Pc: (100) Target Connection Failed
```

gkasprow, 2018-10-12:
please make sure that small golden wire that connects programmer connector with proximity of SCANSTA chip is intact.

gkasprow, 2018-10-12:
Didn't you plug the programmer header rotated?

jbqubit, 2018-10-12:
> Didn't you plug the programmer header rotated?

I noted orientation that worked when Pawel was visiting -- ribbon pointing toward front of AMC. But flipping doesn't help this problem. 

> please make sure that small golden wire that connects programmer connector with proximity of SCANSTA chip is intact.

That wire is intact. 

I see the same error on both of my AMCs. So perhaps its a configuration issue with the flash programmer? I'm using Flash Driver: ```LPC177x_8x_407x_8x_256.cfx``` Is that the correct one?

jbqubit, 2018-10-12:
@wizath Where is the source for the latest version? Looks like path changed in git ```sinara-hw/sinara/SAYMA_firmware/lpc1776_ethernet_I2C``` and [build instructions](https://github.com/sinara-hw/meta/wiki/Sayma#building-openmmc-firmware-for-sayma) no longer work.