sinara-hw/sinara #598: Sayma v2 Ethernet

jbqubit, 2018-11-06:
In Sayma v1 Ethernet is complicated by support for both MMC uP and FPGA. Happily this is now working but it could be made simpler. A low risk simplification is to abandon Ethernet support for MMC and connect MAX24287 directly to FPGA. 

PRO:
- Less complex layout and debugging.
- No need to support both RGMII and MII.
- Can use existing Sayma v1 firmware/gateware. 
- There's no concrete use case at present for using Ethernet on MMC.

CON:
- Any layout change which could introduce new bugs.
- Blocks future use of MMC over Ethernet.

sbourdeauducq, 2018-11-06:
MII does not work with this PHY (rate conversion is not supported, it just outputs garbage with MII at one end and 1000BASE-X on the other) and therefore it is totally incompatible with the MMC.
Ethernet on MMC requires using another PHY which may bring a fresh batch of bugs.

hartytp, 2018-11-06:
What about just scrapping the ethernet phy then and only using Sayma as a DRTIO slave? If we really need Sayma as a master then what about doing ethernet via the SFP like Kasli (needs a gateware port, but that seems like a more productive thing to do than more time spent messing around with these phys).

gkasprow, 2018-11-06:
We don't use MMC Ethernet because the PHY actually do not support speed translation MII -> 1000 base-x mode, even though the datasheet states the opposite. So we already removed the mux.
On other boards including AFCZ I added single chip switch that has both RMII, MII, 2x1000base-x and 1000T. It consumes as much power as standard PHY. The only drawback is MARVELL NDA, but the chip is plug and play, no management is necessary.

jbqubit, 2018-11-06:
> What about just scrapping the ethernet phy then and only using Sayma as a DRTIO slave?

After the long battle to get  1000BASE-X Ethernet working on AMC I don't want to see that capability lost. 

Any objections to connecting MAX24287 directly to FPGA and closing this Issue?

hartytp, 2018-11-06:
> After the long battle to get 1000BASE-X Ethernet working on AMC I don't want to see that capability lost.

Is there an argument for this beyond the sunk cost fallacy?

hartytp, 2018-11-06:
Anyway, I guess that for Sayma this is relatively unimportant anyway. The real question is what we will do on Metlino where ethernet is much more central.

sbourdeauducq, 2018-11-06:
Funnily enough, getting Ethernet to work on Kasli with the transceiver directly and processing the raw 1000BASE-X bits was faster and less frustrating than getting the Sayma PHY and RGMII on Ultrascale to behave.
I wonder how well-tested (i.e. across many boards) the current RGMII solution is. There is the problem of timing/phase calibration and the large I/O timing variations over PVT that Ultrascale is subject to. Improving the clocking (use CC pin in same bank as recommended by Xilinx) should make the situation better though.

jbqubit, 2018-11-06:
> Is there an argument for this beyond the sunk cost fallacy?

Good rhetorical question. :) Maintaining 1000BASE-X Ethernet strikes me as a straightforward way of using the AMC in the absence of Metlino. 
- AFAICT there's little cost to maintaining this capability. The M-Labs build for Sayma v1 as Master already exists. 
- Other projects may make use of Sayma AMC v2 besides SAWG, eg Blaster, and not want to use Metlino. 
- Permits use of two-slot uTCA crate that @gkasprow discussed some time ago. One Sayma as Master, other as Slave.

hartytp, 2018-11-06:
> I wonder how well-tested (i.e. across many boards) the current RGMII solution is. There is the problem of timing/phase calibration and the large I/O timing variations over PVT that Ultrascale is [...]

:+1:

@sbourdeauducq right. Given our positive experience on Kasli, I wonder whether hooking the SFP directly to the FPGA and porting the PHY to Ultrascale would be a better route in the long run. It's not like the current situation is particularly well tested or can really be guaranteed to be "known good".

jbqubit, 2018-11-06:
> I wonder whether hooking the SFP directly to the FPGA and porting the PHY to Ultrascale would be a better route in the long run.

Would this preclude using Sayma as middle-man in SFP-based DRTIO? It only has two SFP ports... 

> It's not like the current situation is particularly well tested or can really be guaranteed to be "known good".

I'd like to minimize new development for Sayma v2. Maintaining "known to work ok" MAX24287 strikes me as better than trying something new with KU7. 

@gkasprow Are there enough of the right type of pins to support the following? Layout so that uTCA Port0 and SATA0 (or uTCA Port0 and SFP) can be driven by either FPGA or MAX24287. Use coupling capacitors to choose which. Default stuffing and development path is MAX24287.

hartytp, 2018-11-06:
> Would this preclude using Sayma as middle-man in SFP-based DRTIO? It only has two SFP ports...

I don't see how it would. AFAICs, ethernet is only used when Sayma is the artiq master. If it's a satellite/repeater node then the ethernet is unused.

There is no hardware difference between SFP-based ethernet and SFP-drtio, just a gateware change; it's exactly the same as Kasli.

gkasprow, 2018-11-06:
For such operation one can always use one of the SFP slots.
In next revision we will place the PHY much closer to FPGA and of course use dedicated clock line.
We can also replace the PHY with popular 88E1512 which is used in many Open Hardware designs. The datasheet was opened to public recently. I have FMC board with 4 such PHYs which work without any issues.

gkasprow, 2018-11-06:
88E1512 has one interesting feature - it can shift the clock itself to meet setup and hold time. It's just one bit in MDIO register. A few days ago we had issues with non-working Ethernet PHY and we figured out that ZynQ US+ requires additional delay on PCB on clock lines. Marvell took it into account and implemented this feature in some of its PHY chips.

jbqubit, 2018-11-06:
If FPGA can stand-in for dedicated Ethernet IC that does seem to have some plusses.
- One fewer item on BOM
- Only hidden demons are those inside Kintex Ultrascale (which we can't really avoid); no more worry about Marvell/Max demons  
- SFP-based Ethernet or DRTIO
- If this SFP were tied to uTCA Port 0 pins then 1000BASE-X could use either SFP or backplane. 

Why did we ever bother with an Ethernet IC in the first place? I guess that decision predated the positive experience with Kasli.

sbourdeauducq, 2018-11-07:
> 88E1512 has one interesting feature - it can shift the clock itself to meet setup and hold time. It's just one bit in MDIO register.

All this does is move one PLL/MMCM from the FPGA into the PHY chip, and reduce its resolution drastically (reading the datasheet, there are only two possible delays!).
Does this feature even work correctly or is it more like something that HMC would have designed?

> A few days ago we had issues with non-working Ethernet PHY and we figured out that ZynQ US+ requires additional delay on PCB on clock lines.

This is one of the many reasons I don't like Xilinx IP. With open fabric logic it is quite straightforward to add clock shifting capability.

> Why did we ever bother with an Ethernet IC in the first place?

* No past experience as you mention and also one has to deal with the GT* transceiver (which isn't straightforward).
* I wanted the same circuit as the KC705 since that was proven to work and has quite relaxed timing and simple implementation on the FPGA side.
* Greg wanted Ethernet on MMC.

sbourdeauducq, 2018-11-07:
> Layout so that uTCA Port0 and SATA0 (or uTCA Port0 and SFP) can be driven by either FPGA or MAX24287. Use coupling capacitors to choose which.

Sounds like a reasonable risk mitigation plan.

jbqubit, 2018-11-07:
@sbourdeauducq In net, how do you feel about using Kasli approach on Sayma v2? Is this a day of work or more?

sbourdeauducq, 2018-11-07:
It's more. It would be easy if Xilinx made transceivers compatible across FPGA families, but they definitely did not.

jbqubit, 2018-11-07:
OK. Sounds like we have a satisfactory plan to move forward with for Sayma v2. 

> Layout so that uTCA Port0 and SFP can be driven by either FPGA or MAX24287. Use coupling capacitors to choose which. Default stuffing and development path is MAX24287.

Closing Issue.

hartytp, 2018-11-07:
> If FPGA can stand-in for dedicated Ethernet IC that does seem to have some plusses.

Exactly. What concerns me about the current ethernet implementation (and, indeed, what actually caused problems and lead to huge amounts of wasted time) was the complex stack involved here. We have a somewhat fussy ethernet phy with a lot of features we don't need and a data sheet that runs to hundreds of pages. It needs to be configured by MMC firmware that's not maintained by M-Labs (and, despite good work from Greg's team, fixes to the MMC firmware have been frustratingly slow to emerge) and then needs a gateware interface. This means that if anything goes wrong then it's hard to fix since there is no single person who has ownership of the whole system.

For better or worse, if we go for SPF directly to transceiver then M-Labs are responsible for the entire system so it's clear who to speak to if there is a problem. From a project management point of view, that's extremely valuable.

> Layout so that uTCA Port0 and SATA0 (or uTCA Port0 and SFP) can be driven by either FPGA or MAX24287. Use coupling capacitors to choose which.

That sounds like a decent plan. As always, the hard question will be what to do if we do find ethernet issues in Sayma v2.0. Do we spend "just a bit more time" fixing the current implementation that "almost works" or do we switch to developing a new solution.

---

Anyway, this seems like something that is more pressing for metlino than Sayma, since ethernet is more crucial there.

gkasprow, 2018-11-07:
we can also use identical PHY as KC705 uses. It does not cause any issues and also supports 1000base-X.
M88E1111 has documentation available.

jbqubit, 2018-11-07:
> [MAX24287] needs to be configured by MMC firmware that's not maintained by M-Labs

Good point @hartytp. Putting Ethernet into the hands of a single developer is quite desirable. I'd prefer that the Software and Gateware Developer do it. M-Labs prefers M88E1111. And using same IC as KC705 (M88E1111 ) means the firmware should be more or less identical. 

New proposed wording for Ethernet part of contract. 

- Replace MAX24287 by M88E1111. This is same IC as KC705 which is already supported by ARTIQ. 
- FPGA configures M88E1111 -- no interface with MMC. 
- Layout so that uTCA Port0 and SFP can be driven by M88E1111 or by GTH transceiver on FPGA. Use coupling capacitors to choose which.
- Default stuffing and development path is M88E1111. GTH path is for risk mitigation.

gkasprow, 2018-11-07:
the same applies here https://github.com/sinara-hw/Metlino/issues/3

sbourdeauducq, 2018-11-08:
NB: On the KC705 we are not doing any configuration of the PHY. It is already set up (probably via pin-strapping) to establish automatically a 100M or 1G link on the RJ45, and present a MII (if the link was 100M) or GMII (if the link was 1G) interface to the FPGA. The FPGA determines which interface it is by measuring the frequency of the clock sent by the PHY. We never touch the PHY registers.

sbourdeauducq, 2018-11-08:
As Greg pointed out in the RFSoC issue, the regular Ultrascale I/O pins can in theory be used for 1000BASE-X directly; however GTH is preferable for those reasons:
* silicon design and wizard code are less bad
* past experience with Kasli (parts of GTH and GTP are similar)
* compatibility with SFP ports that are also usable for DRTIO

gkasprow, 2018-11-20:
@sbourdeauducq 88E1111 is hard to get from regular sources. It's very old chip. People use 88E1512 instead. I used it already without issues. It is the one installed on [FMC-Ethernet](https://ethernetfmc.com/) board. Any objection ?

gkasprow, 2018-11-20:
88E1512 has documentation available without NDA. We will use it also on new simple SAMC-FMC which we design with Kicad.

sbourdeauducq, 2018-11-20:
>  Any objection ?

It's RGMII again. Why not keep the MAX chip then, since it's already debugged?

gkasprow, 2018-11-20:
I don't trust this MAX chip. I never saw anybody using it. There can be another issues.

gkasprow, 2018-11-20:
The new PHY is much simpler, simple supply, no external oscillator and I used it already.
[image]

closed 2018-11-27 by gkasprow