sinara-hw/sinara #599: Sayma v2 design tool

gkasprow, 2018-11-06:
Recently I moved several designs from Xpedition to Altium Designer.
64-bit AD18 is far more stable than AD17. I already designed a few complex boards with it.
One year ago it was nearly impossible to handle such design with AD17. It crashed once every a few minutes and one had to carefully observe amount of allocated memory to not exceed roughly 1.5GB.

I quickly ported schematics and layout of Sayma RTM to AD18, but of course it needs roughly 2 days to fully synchronize designs. The overall quality is very good.  This is possible because I used Altium libraries for Sayma design with Xpedition so can quickly replace converted libs with native ones.
We have commercial Xpedition license at the WUT as well as AD18 one so can continue with any of the tools.
The question is - how important is selection of design tool? Would such design be more attractive if it was done with AD instead of Xpedition? It of course add a risk that translation and manual fixing adds some critical bug, that's why I ask if that makes sense.

AD has much better schematic editor, library management and documentation generation tool. Xpedition has much better PCB editor but has very poor documentation generation and 3D tools.

Recently, as a kind of experiment, we started design of AMC-FMC board (complexity of Kasli) with Kicad for one of US labs. It takes roughly 2x more time than with AD and the tool lacks several critical features but guys at CERN are doing great job. There is no straight conversion between AD/Xpedition and Kicad so one have to start  nearly from scratch. Already available conversion tools are good for very simple use cases.

dhslichter, 2018-11-06:
@gkasprow I vote that we don't change horses mid-stream -- stick with Xpedition to finish out Sayma, and don't risk errors and other complications from porting.  I agree that AD has advantages, but let's save it for new projects we are starting from scratch.  

I think that there is no comparison between Altium Designer/Xpedition and KiCad, and that for large-scale professional projects it makes sense to use AD or Xpedition, especially if you already have had the experience that KiCad takes longer due to lack of critical features.  KiCad is a nice tool for smaller projects and has the advantage of being free, but for the Sinara ecosystem application it seems we want a more powerful option.

marmeladapk, 2018-11-06:
Schematics from Altium are also much more readable, fit in repository (see AFCZ) and with available viewer it's easier to traverse schematics and find bugs in projects (and @hartytp is quite skilled in that ;) ). Also guys at Creotech would much appreciate it. And it lowers the entry barrier if someone would want to contribute.

@gkasprow is it possible to port Sayma v.2 to Altium once PCB routing is done and also import PCB?

gkasprow, 2018-11-06:
@marmeladapk Yes, this is what I did already. You can import existing design with AD18 and treat AD as a viewer :) Full synchronisation takes some time. For single-sheet design it is one hour of work. For complex designs one have to replace all buses with harnesses and this takes a lot of time.

gkasprow, 2018-11-06:
On the other hand, every AD user can do the same - import the Xpedition design and continue development with AD tool.

jbqubit, 2018-11-07:
I can see how using AD for this project has many plusses. However, given present need for low risk, quick turn around in Sayma v2 I don't think now is the right time to tackling porting to a new tool chain.

gkasprow, 2018-11-21:
btw, I split the Xpedition project to 3 different pojects. WIth new release of Xpdedition the GUI became very unresponsive - the database is simply too big and I had to wait 10s after each edit. After I split to Metlino, Sayma AMC and Sayma RTM the schematic editor resonds much better, i wait only 2s :D

closed 2018-11-21 by gkasprow