sinara-hw/sinara #600: Sayma v2 design changes

gkasprow, 2018-11-20:
In this issue I will document all rework that was done in rev1 and how they were applied in rev2. It also documents changes to the design.

gkasprow, 2018-11-20:
Slow start circuit for FMC 12V
new schematics - gray symbols were added
[image]

[image]

gkasprow, 2018-11-20:
RTM slow start
[image]
[image]

gkasprow, 2018-11-20:
similar slow start ciecuit was added for RTM P3V3_MP 
[image]

gkasprow, 2018-11-20:
Main Payload power current limit was added. Before it was just 0.02R resistor
[image]

[image]

gkasprow, 2018-11-20:
Exar chip GPIO0 and PSIO2 pins were swapped. GPIO pins are push-pull while PSIO pins are open drain. So PSIO is better suited as input and GPIO as output.
[image]

[image]

gkasprow, 2018-11-20:
Missing pulldown resistors for LVPECL outputs were added. LVDS version can be used as well but its availability is worse.
[image]

[image]

gkasprow, 2018-11-20:
swapped IO direction control was fixed
[image]

[image]

gkasprow, 2018-11-20:
Too low VCCINT voltage, drop during high current consumption. Rfb selected to give 0.95V. R508 installed to compensate voltage drop.
[image]
[image]

gkasprow, 2018-11-20:
Ethernet issues:

[image]
[image]

need radical solutions.
First, correct RX_CLK pin on FPGA. We won't need TCLKC_D so can use this GC for RX clock.

[image]

Second, disconnect from MMC chip. There are no PHYs that support 100M<-> 1G speed translation.
So we will stick with popular, available and well known 88E1512.
The MMC will be connected only to MDIO bus, just in case. ETH1_MII name is changed to PHY_CFG

[image]

We also get rid of the Mux and PHY schematic gets simplified a lot.

gkasprow, 2018-11-20:
The new PHY is much simpler, simple supply, no external oscillator and I used it already.
[image]

gkasprow, 2018-11-20:
MMC reset during USB plugging was fixed by adding RCD circuit. It was verified on AFCZ board.
[image]
Flashmagic boorloader mode can be disabled using DIP switch
[image]

It was tested on AFCZ

before wire jumper was used

[image]

gkasprow, 2018-11-20:
MMC JTAG wac blocked by SCANSTA. RCT circuit was added to disable SCANSTA chip when the MMC programmer plug is inserted.

[image]

[image]

Analog switch was added that cuts SCANSTA JTAG access when the JTAG plug is inserted. It was verified on AFCZ 

[image]

gkasprow, 2018-11-20:
LTC6957 had SD pins tied to VCC instead of GND
[image]
[image]

SMA clock output was connected to LTC6957 output and recovered clocks delivered to FPGA had phase shift. IT was fixed by connecting SMA output to the Si5324s chip directly.

sbourdeauducq, 2018-11-21:
This SCANSTA caused a relatively low number of problems (by Sayma standards), but do we really need it?

hartytp, 2018-11-21:
@gkasprow which change fixes USB JTAG so that no jumper is needed?

gkasprow, 2018-11-21:
Instead of SCANSTA we would have to use discrete JTAG muses and potentially add issues. @hartytp This was fixed by another line routed to MMC so it knows state of the USB JTAG and resets SCANSTA. I will explain it later on.

gkasprow, 2018-11-21:
If we are going to support remote update over RTIO, I can remove the SCANSTA entirely. The switches can be also controlled via MMC as it does with SCANSTA.

sbourdeauducq, 2018-11-21:
Currently there is no funding for DRTIO firmware updates, but there is also no funding for µTCA firmware updates. The former is much easier and lower-cost to implement, and would also support Kasli.

gkasprow, 2018-11-21:
uTC firmware update is performed by NAT MCH + Jtag Switch Module. At least my coleagues in DESY use it in such way. Personally I never tested it. I used Vadatech JSM so far and that's why implemented switches.
But we can keep access to the AMC JTAG using muxes, without using SCANSTA chip.
On AFCZ board I added SCANSTA in both BGA and TQFP package because had issues with availability.
It can be done in the same way as on KC705, with ability to overdrive FMC and RTM detect pins by MMC. AMC access can be performed by additional mux. In this way we keep simplicity. We loose ability to control JTAG chain via JTAG commands which worked quite nicely in AFCK board, but once we are able to force working configuration with MMC, I don't care.

gkasprow, 2018-11-26:
The RTM FPGA config pins were connected 0 0 1, it should be 1 1 1.
[image]
Now we have 
[image]

gkasprow, 2018-11-26:
RTM FPGA config pins were swapped. DIN is not data in but D00_MOSI.
HW fix:
[image]
[image]

Schematic fix:
[image]

gkasprow, 2018-11-26:
Missing connection of VTT with SVDD12
[image]
[image]

fix:
[image]

gkasprow, 2018-11-26:
Broken feedback loop for DC/DC converters
[image]

The VFB resistors must go to GND.

[image]

gkasprow, 2018-11-26:
IC34 SD pins should be grounded.
[image]
[image]

gkasprow, 2018-11-26:
not enouth current for SVDD12
[image]
[image]

fix - bridge 2 LDOs together with 20mOhm resistors
[image]

gkasprow, 2018-11-26:
Fix power sequencing of negative SEPIC converter -  by addin 1uF cap to SS to make sure negative converter starts after positive one.
[image]
fix
[image]

gkasprow, 2018-11-27:
Missing DAC output termination and input clock termination
[image]

Added 100R terminators to SYSREF and CLK and 49R9 to outputs.
[image]

gkasprow, 2018-11-27:
add 10k pulldown on PLL chip CLK and SEN lines to make sure that logic state is well defined and FPGA can setup correct SPI protocol.
[image]
fix
[image]

gkasprow, 2018-11-27:
Wrong schematic symbol of interdigitated capacitors
[image]

Caps had to be soldered one side only.
Pins 5-8 were connected in reversed order
fix:
[image]

sbourdeauducq, 2018-11-27:
> HMC7043 not connected RSV lines,

Thankfully, we're not using this chip anymore.

gkasprow, 2018-11-27:
So, will we switch to HMC7043?

hartytp, 2018-11-27:
> So, will we switch to HMC7044?

No, we will remove the HMC7043 entirely and replace it with a fanout + DFF + delay line. 

See here: https://github.com/sinara-hw/sinara/issues/594

hartytp, 2018-11-27:
NB there is a list of changes for Sayma v2.0 here https://github.com/sinara-hw/sinara/milestone/22

gkasprow, 2018-11-27:
Daisychain FPGA configuration did not work for unknown reason, so we switched to direct config of RTM FPGA by AMC FPGA. Resistors must replaced:
[image]
[image]

So we make daisychain connection DNP, we can also remove it later on
[image]

and we mount
[image]

gkasprow, 2018-11-27:
SCANSTA chip blocks MMC JTAG even when port is inactive. The simplest way was to disable the SCANSTA chip entirely when MMC JTAG plug was detected. 2x 10k resistors, 100nF capacitor + MOSFET are needed.
[image]
The SCANSTA_DISABLE goes instead of R29
[image]

gkasprow, 2018-11-27:
SCANSTA fails to switch to internal mode when USB is plugged after board was powered on. Simple solution is jumper on cable detect line.
[image]

On rev 2 new JTAG switching scheme will be used.

hartytp, 2019-03-06:
Closing as I don't think there will be any new rework for Sayma v1.0

closed 2019-03-06 by hartytp