sinara-hw/sinara #601: Sayma v2 Project Plan

jbqubit, 2018-11-29:
Here's a coherent project plan for Sayma v2. It is informed by extensive discussion on github Issue system over the past year and recent offline discussion with the Sayma v1 developers. This is posted here so everyone knows what the contracts call for in development of Sayma v2. This plan will be updated if contracts differ from the present plan. Please use existing or new Issues for further discussion of technical topics. 

This plan defines concrete performance goals, specifies which parties are responsible for specific deliverables and defines a project management structure. The aim is to provide sufficient specification and structure at the outset that all parties feel comfortable with their responsibilities and minimizes the role of ARL/UMD in project management. 

The plan includes design choices that simplify Sayma and lower costs. The github Sayma Issue tracking contains a long list of bug fixes and modifications to v1 (https://github.com/sinara-hw/sinara). I call out explicit specifications below in cases where there is not a clear github consensus on how to proceed with Sayma v2. The plan is not meant to substitute for the Issue system except where which-path clarity is required in which case this plan is authoritative. I will annotate relevant github Issues with a link to this Issue so there is clarity about which-path.

# Definitions
Let __TS-MTCA__ be a micro TCA crate configured as follows: 
- 1 NAT NATIVE-R5 or -R9
- 1 NAT AC 600D 
- 1 NAT MCH-Basic v3.5

Let __TS1__ be a Sayma test setup consisting of the following
- 1 Sayma AMC+RTM 
- Powered by ATX connector from benchtop supply

Let __TS2__ be a Sayma test setup consisting of the following
- Single uTCA chassis configured as TS-MTCA
- 2 Sayma AMC+RTM
- 1 Metlino 
- All RF channels producing 400 MHz sinusoid
- Cards in adjacent slots

Let __TS7__ be a Sayma test setup consisting of the following
- Single uTCA chassis configured as TS-MTCA
- 6 Sayma AMC+RTM
- 1 Metlino 
- All RF channels producing 400 MHz sinusoid
- Cards in adjacent slots

# Sayma v2 Diffs
When diff also applies for [Metlino](https://github.com/sinara-hw/Metlino/issues) it is noted. 

## RTM Synthesizer
- Layout both ADF4351 and HMC830, change capacitors to reroute clocks.
- Use HMC830 as primary synthesizer. 
    - problem: random lockups at power cycling, possible fix: possibly due to power sequencing 
    - problem: unsynchronised dividers, fix: Make FPGA measure the output phase and reset the HMC830 divider until synchronised.
    - HMC830 lock status exposed on front-panel LED. 
- ADF4351 is backup option. No development of ADF4351 until operation with HMC830 is fully tested and succeeds, or HMC830 bug clearly blocks forward progress. 

## AMC clock recovery
Discussed in [meta/issues/15](https://github.com/sinara-hw/meta/issues/15). Also applies to [Metlino](https://github.com/sinara-hw/Metlino/issues).
- Use Si5324 as primary approach for Sayma v2.
- Layout both Si5324 and Tom’s WR PLL scheme with Si549
    - Change capacitors to reroute clocks.
    - Confirm that DCXO and the Si5324 can both be shutdown in software (using SFOUTx_REG), otherwise add a FET on power lines.
- Clear delineation on schematics between Si5324 and WR PLL schemes. 
- No funding to develop WR PLL support in this contract.

## DAC-FPGA Synchronization
Deterministic phase alignment between DACs on separate Sayma PCBs between power cycles is an important design criterion and should be considered at all stages of design. This implies a) SYSREF is aligned with RTIO clock b) DAC clocks are synchronized to local SYSREF edges. 
- Follow approach demonstrated by Tom's prototype (Issue #583) using low jitter FF to retime AMC-FPGA SYNC pulse to f_dac_ref. Summarized in Issue #594. 
- Remove HMC7043, add components needed to generate MGTREFCLK. 
- Test conditions
    - SAWG output on all channels on two Sayma boards, 400 MHz continuous RF 
    - Sayma-local TTL output at 100 MHz
    - laboratory temperature stability +/- 0.5 C
    - Cycle crate power 100 times
- Specification 
    - __ST1__: SAWG-SAWG phase alignment error across 
all power cycles: < 1/f_dac_clk 
    - __ST2__: SAWG-TTL alignment error across all power cycles: < 0.5 ns

## DAC Performance

### DAC Phase Stability
Phase stability between DACs on separate Sayma PCBs is an important design criterion and should be considered at all stages of the design. This assumes that DAC-FPGA synchronization.  
- Test conditions
    - SAWG output on all channels on two Sayma boards, 400 MHz continuous RF 
[... 214 lines cut ...]
- __M34__ removed
- __M35__ removed 
- __M36__ build infrastructure 
    - Support build for bitstream building for Sayma v1 and SAWG v1 until 6 months after HT3 delivery. 
    - Support for bitstream building for SAWG v1 on Sayma v2 until 6 months after SAWG v2 delivery. 
    - Support for bitstream building of SAWG v2 for Sayma v2 for 3 years after SAWG v2 delivery. 
- __M37__ support reduction of data rate for FTDI chip -- pending

----------

# Project Organization
ARL/UMD will fund and manage contracts with @jbqubit serving as technical point of contact (TPOC). In the event of a dispute about interpretation of a contract or about contract execution, the TPOC will have project-level decision making. Note that the legal language in the officially executed contract documents has final authority. 

New github repositories were setup for each component of the Sayma system. 
- https://github.com/sinara-hw/Sayma_AMC
- https://github.com/sinara-hw/Sayma_RTM
- https://github.com/sinara-hw/Metlino
- https://github.com/sinara-hw/BaseMod

Github Issue tracking will be the authoritative location for test results and bugs. Keep discussion via other channels (email, IRC) to a minimum but when it happens post a summary to the relevant github Issue. Link back to the old sinara-hw/sinara Issues as needed but don’t conduct new discussions in sinara-hw/sinara. 

Weekly discussions via google hangout during period of performance. Attendance of at least one team member for each collaborator is expected. Discussion limited to 30 minutes per week. Software and Gateware Developer, Hardware Developer and Integration and Timing Developer. UMD/ARL involvement if requested by MC. Responsibilities:
- Strict cycling of responsibilities between teams for each week
- Master of ceremonies (MC): publishes meeting agenda at least 4 hours prior to meeting start, enforces time limits, schedules next weekly meeting (advertised in UTC)
- Clerk: takes meeting minutes, posts to github

hartytp, 2018-11-29:
> problem: unsynchronised dividers, fix: Make FPGA measure the output phase and reset the HMC830 divider until synchronised.

In the first instance, the fix should be to not use the HMC830 output dividers, and just run the DAC at > 1.5GSPs using interpolation.

Using the FPGA to synchronise the PLL output dividers is mainly a risk mitigation strategy in case there is some issue with interpolation (e.g. DAC synchronisation doesn't work at max clock rate due to a silicon bug).

hartytp, 2018-11-30:
> f_dac_clk = 2.0 GHz (4X interpolation) with following expected implications:

Typo, should be 2x.

hartytp, 2018-11-30:
> BaseMod v2: print 20 PCBs, stuff 12

We should aim to distribute test AFEs as well which just run the DAC outputs to SMAs via baluns. This gives a path to testing DAC issues without having to configure RF switches, attenuators etc, which might be helpful in initial debugging.

sbourdeauducq, 2018-11-30:
> Using the FPGA to synchronise the PLL output dividers is mainly a risk mitigation strategy in case there is some issue with interpolation

I would prioritize getting the alternative chip to work over this.

jbqubit, 2018-12-12:
@hartytp Project Plan addition: added to __HT3__ ```TestMod v2: print 6 PCBs, stuff 6 (DAC outputs to SMAs via baluns)```

@hartytp Project Plan addition: __O18__ Integration and Timing Developer responsible for writing and testing ADF4356 driver. 

>> M32 support Si549 as stand-in replacement for Si5324 (no White Rabbit time synchronization in this contract)

> What's this one about? The SI549 isn't a replacement for Si5324 without WR.

Project Plan modification: M32 is now crossed off since WR support is not covered by the plan.

jbqubit, 2018-12-12:
Project plan modifications:

M12 design study to support 1 GSPS data rate on Sayma v2 ---> M12 design study to support 1 GSPS data rate on Sayma v2 (SAWG v1)

M13 design study to investigate SAWG v2 ---> M13 design study to investigate SAWG v2 (eg limit f_f0 resolution, reduce RTIO channel number, Moninj) 

 __M15__ lead HT1 design review ---> __M15__ participate in HT1 design review 

__M24__ test AFE BaseMod v2 (excluding ADC) ---> __M24__ test AFE BaseMod v2 (including RF switches and attenuators, excluding ADC) 

Addition of new task: M28 extend RTIO support to FPGA on RTM

__M31__ ---> __M31__ support code developed in __O18__

__M37__ support reduction of data rate for FTDI chip ---> removed 

Addition of new task to __HT3__: 
- Metlino AMC: print 10 PCBs, stuff 4

--------------

__HT2__ design review modifications 

- Strike M16. 
- Modify M17 to read "participate in HT2 design review"
- Addition for Hardware Developer
Hardware Developer shall lead HT2 design review. This design review is in collaboration with scientists and other developers. The review will not be comprehensive. It will focus on differences between v1 and v2 of the hardware. Shall include validation of FPGA pinout.

jbqubit, 2018-12-12:
Minutes for last two discussions: 
- https://github.com/sinara-hw/sinara/wiki/Minutes20181204
- https://github.com/sinara-hw/sinara/wiki/Minutes20181211-Sayma-v2

jbqubit, 2018-12-12:
Modifications to Project Plan in light of email from Creotech.

__TS-MTCA__ modifications
- 1 NAT NATIVE-R5 or -R9 ---> 1 NAT NATIVE-R9 
- 1 NAT AC 600D  ---> 2 NAT AC 600D 

__HT3__ modification
- thermal test in TS7, Maximum temperature of any IC 65 C ---> thermal test in TS7, Maximum temperature of any IC 65 C, monitor temperature of several representative ICs using MMC IPMI 

__HT4__ modification of formal hardware documentation maintained in github 
- scope: expected content is superset of existing Sayma v1 documentation
- documentation to also include the following github Issues
https://github.com/sinara-hw/sinara/issues?q=is%3Aissue+documentation+label%3Aarea%3Adocumentation+is%3Aopen

Comments on question in Creotech email.

>Should the ARTIQ Python code tests belonging to HT5 really belong to the tasks of the Hardware Developer?

Yes. It is crucial that the Hardware Developer demonstrate expertise in using ARTIQ and Sinara together. Sayma v1 testing was handicapped by not having such expertise in Warsaw. In particular, there were hardware bugs that didn't emerge until the full ARTIQ system was running on the Sayma v1 hardware. 

> It seems that these tests depend more on the status of GW/SW than the hardware. If problems with GW/SW arise, how should they be addressed?

I understand that GW/SW problems may cause problems for delivery of HT5. Indeed, if hardware problems arise it is becomes difficult for the Integration and Timing and Software/Firmware Developers to deliver certain contract items. For Sayma v1 there was no contract item that made Warsaw dependent on GW/SW Developers. Now it is explicitly built-in to reflect the co-dependencies of the project. In contract bids, I anticipate that each of HT1 to HT6 will be a separate line item. Each line item can be billed for independently. So, if HT5 is slowed due to GW/SW problems, the remaining HTx could still be billed.

jbqubit, 2018-12-12:
@hartytp asked

> who is going to lead the design review? By "lead", I mean: drawing up a list of things that need review (e.g. based on the CERN check list); organising who will be responsible for reviewing each aspect of the design; chasing people up to ensure that they perform their parts of the design review and sign off on the design; having the final sign-off on the project before it goes to manufacture. This needs to be done by someone who is following the technical aspects of the project closely. 

Creotech will manage the design review. As a starting point @marmeladapk and @jbqubit are taking a list of v1 to v2 diffs supplied by @gkasprow and generating a list of specific areas that need review. We are also making an explicit list of checks based on CERN checklist. Will be sent out as RFQ tomorrow. I'll add @hartytp to the list of editors now.

marmeladapk, 2018-12-13:
@hartytp If you check design changes table, which Joe shared with you, you should see that all open issues from milestone are added there. I'm going now through all open issues with no milestone and adding them if necessary. Then I guess I should also check closed issues, though we could share the workload if you'd be willing to.

> how do you want to handle these issues? Do you want to leave them open until someone has done an independent design review and signed off that the changes were correctly implemented? 

I think that person that has been assigned to check an issue should give a green light to close it. Another way is to ask @gkasprow to close all issues implemented, that are in Sayma v2.0 milestone and rely solely on design review table, that we are making right now.

> If so, someone should also do the thankless job of checking that all the closed issues really have been implemented.

That's true, but most of them should be obsolete.

marmeladapk, 2018-12-13:
I went through:
- closed milestone issues,
- open and closed issues with no label,
- open and closed issues with sayma amc or sayma rtm label,
- AMC and RTM errata issues.

jbqubit, 2018-12-13:
From https://github.com/sinara-hw/sinara/issues/602#issuecomment-446916631 I've changed the following. 

ADF4351 --- > ADF4356

jbqubit, 2018-12-19:
@hartytp pointed out that his synchronization patch O11 only relates to ST4. Software and Gateware Developer are best situated to handle ST1 and ST2 which depend on DRTIO. So some changes...

> M23 implement and test ST2 and ST4. Software and Gateware Developer

--> M23 implement, test and document ST1 and ST2  which depend on DRTIO; add to continuous integration

> O11 patch to implement DAC-FPGA synchronization

--> O11 implement DAC-FPGA synchronization with 2 GHz DAC clock
(1 GSPS data rate).

> O14 responsible for building, conducting and documenting tests for ST1 and ST3

--> 
O14  Based on O11, demonstrate ST4 for a pair of AD9154 DACs on a single Sayma v2 board
in configuration TS1.
- Write and document code. Generate pull request.
- Re-test the ARTIQ-integrated version of the code and debug any
issues that arise.
- Dependency: Software and Gateware Developer is responsible for
supplying AD9154 support at the level of M25 and M32. Integration and
Timing Developer will handle debugging of ADF4356 and AD9154 as relates to ST4.

jbqubit, 2018-12-19:
> O15 measure DAC phase stability
> O16 measure DAC phase noise

O15 and O16 moved out of Project Plan and assigned to individuals in github. 
https://github.com/sinara-hw/Sayma_RTM/issues/1
https://github.com/sinara-hw/Sayma_RTM/issues/2

jbqubit, 2018-12-26:
Edit to make explicit a change that will prevent https://github.com/sinara-hw/sinara/issues/571. Heads up @marmeladapk. 

Add to HT3:
- For TS-MTCA with NATIVE-R5, publish MCH configuration configuration file used for these tests.

jbqubit, 2018-12-26:
Modification to MT1 based on email with @sbourdeauducq.
* M17 Replace serwb with pure DRTIO.
* RTM DRTIO -- already funded, can be prototyped on Sayma v1 or Kasli
* distributed DMA -- already funded, can be prototyped on Sayma v1 or Kasli

hartytp, 2018-12-26:
> RTM DRTIO -- already funded, can be prototyped on Sayma v1 or Kasli

What does prototyping this involve? IIRC, it's already prototyped in the sense that DRTIO between Sayma AMC and Kasli already works, but we can't directly prototype DRTIO with the RTM FPGA on v1.0 HW since there are no AMC<->RTM transceiver links...

sbourdeauducq, 2018-12-27:
Use a SATA cable. I'll work on that after I get back from Xmas holidays.

hartytp, 2018-12-27:
Ace, thanks!

jbqubit, 2019-01-18:
Everyone, please review a draft update to the project plan at the following URL. 

https://github.com/jbqubit/saymapp

The number of changes has gotten numerous enough that git is useful for tracking. The present revision reflects feedback from all prospective Developers. The jbqubit repository is temporary; once everyone agrees upon changes I'll update this Issue.

hartytp, 2019-03-11:
@jbqubit do we still need to keep this issue open, or can we close it?

hartytp, 2019-03-12:
@gkasprow can you let me know when the amc and rtm schematics are ready for review?

gkasprow, 2019-03-12:
Yesterday I published a new release for both of them.

hartytp, 2019-03-12:
Okay, thanks.

I didn't start reviewing yet as there are still some outstanding issues. If you confirm that all schematic issues (apart from finalizing the HMC830 loop filter OpAmp) are resolved then I will begin my review.

hartytp, 2019-03-12:
@jbqubit @marmeladapk can you confirm how you want us to structure the review/who is responsible for what?

gkasprow, 2019-03-12:
Yes, I confirm.

hartytp, 2019-03-13:
Thanks Greg.

@jbqubit please can you confirm how you want us to do the review. AFAICT the google spreadsheet you circulated before is now out of date. Is there anything in particular you need me to do? Otherwise I'll review the bits I feel qualified to look at and leave the rest.

hartytp, 2019-03-13:
I've completed my review of Sayma AMC. I didn't find anything major. I'll aim to finish the RTM tomorrow.

jbqubit, 2019-03-15:
@hartytp Thanks for doing the review of AMC. It will help others to know exactly what you reviewed. Please update the spreadsheet to reflect what you checked. Make modifications to the columns if there's something that warranted review but wasn't forseen in the review structure.

@sbourdeauducq Is your AMC stub port complete? Can you articulate what aspects were covered by that port using the google spreadsheet? If not please suggest an alternate format for communicating what aspects of the design you checked.

hartytp, 2019-03-15:
> @hartytp Thanks for doing the review of AMC. It will help others to know exactly what you reviewed. 

Yes, but it would have helped me if you'd responded to any of the times when I asked you to confirm what you actually wanted me to do *before* I completed the review...