sinara-hw/sinara #604: Sayma v2 Design Review

jbqubit, 2018-12-14:
@marmeladapk and @jbqubit put together a spreadsheet that details the design review process for Sayma v2. It includes the HT1 and HT2 review stages. It calls out detailed topics to review, a detailed checklist, assignments and deadlines. Named reviewers will be provided with write access using their google credentials. Please read instructions. Note that the layout design questions are still being refined. Here's a read-only link. 

https://docs.google.com/spreadsheets/d/1tiuDPXyfGAkjgAhQeABbQPNrsW7kpbcXJXQC5EBzqfM/edit?usp=sharing

If an assignment is problematic or you'd like to see another review topic please discuss in this Issue. 

@sbourdeauducq @jordens @dhslichter @hartytp @gkasprow @tprzywoz

jordens, 2018-12-14:
I get access denied.

klickverbot, 2018-12-14:
(That link seems to require credentials as well.)

marmeladapk, 2018-12-14:
Link updated. @jordens @klickverbot

jordens, 2018-12-15:
It's going to be inefficient and error-prone to split tasks like the "all unused inputs terminated" and "diode polarity" checking among 4-6 different groups and people. If you want to split the review, I'd e.g. rather have someone do schematic reviews at the fundamental level (diodes and inputs) and someone else with expertise in different conceptual areas cover bus logic, transcievers lanes and clocking networks.
The person checking some sheet/area (like FPGA decoupling) on Metlino should be whoever did or does it on Sayma.

marmeladapk, 2018-12-17:
@jordens The idea was, that each person has only too look and check as few pages of schematics and data sheets as possible. If someone is checking if issues are properly fixed he can also look at N/C pins, inputs etc.

jbqubit, 2018-12-17:
@jordens The document that @marmeladapk and I put together is structured to make sure to ensure that the review is comprehensive across the Sayma v1 to v2 diffs.  Indeed there are many ways to scale this fish. Given the time available I'd like you and @sbourdeauducq to divvy up the design review tasks assigned to M-Labs. If you prefer to conduct a lateral review of a particular aspect (eg diodes and inputs) across all schematics please call that shot in this Issue and annotate the corresponding rows in the spreadsheet indicating that you've got that aspect covered. Everyone will appreciate your contributions to the review.

jordens, 2018-12-17:
Let's have M-Labs focus on the areas and aspects that we are most qualified to review to the best use of the available funding.

jbqubit, 2018-12-18:
@jordens I appreciate your comment but it doesn't help move the design review forward. Working from the current state of the HT1.5 and HT2 spreadsheet, do you agree that the roles assigned to M-Labs are within the qualifications of M-Labs to review?

jbqubit, 2018-12-18:
@sbourdeauducq Raised a question in [Issue](https://github.com/sinara-hw/sinara/issues/566#issuecomment-448326177). He is recalling the following outcome of a discussion with Xilinx tech support related to Sayma v2 (April 2018).

> Conclusion of Xilinx service request (10440122).  Vccint + Vccint_IO rail should be 0.95 V. Xilinx use of HyperLynx modeling tool. Found that at 9.8A there is a max voltage drop of 105.8mV. Which is too much sag. 

Does the Creotech secondary design review include repeat of signal integrity and power analysis simulations? Do we want this?

jbqubit, 2018-12-18:
https://github.com/sinara-hw/Sayma_AMC/issues/6#issuecomment-448186699

@jordens @sbourdeauducq Funds are on hand now at M-Labs that can be used to start the design review. I've asked several times for a quote to provide additional funding. We all really value your participation in this process. Please don't imply that lack of funding is an obstacle to your participation in the design review.

hartytp, 2018-12-18:
@jbqubit I think the comment about funding there replied to funding for servo gateware/software development, not to general funding for Sayma v2.0 design reviews. AFAICT there is no plan to fund servo development, so that comment is valid.

jbqubit, 2018-12-18:
> AFAICT there is no plan to fund servo development, so that comment is valid.

Indeed, there is presently no formal support in place for Sayma v2 servo development. As I understand https://github.com/sinara-hw/Sayma_AMC/issues/6 is to equip the hardware with sufficient IO to support a possible future servo. Apologies if I misunderstood the scope of @jordens comment.

hartytp, 2018-12-18:
> As I understand sinara-hw/Sayma_AMC#6 is to equip the hardware with sufficient IO to support a possible future servo. 

That's correct. However, without first discussing detailed requirements and then designing the servo in detail, it's hard to guarantee that this will actually work as intended. With the funding and time available, we've done the best we can, and checked the obvious things. It will probably work fine.

sbourdeauducq, 2018-12-19:
>  I've asked several times for a quote to provide additional funding.

You kept asking during the course of about a week; sorry, but we need more time, especially with all the things that can go wrong.

hartytp, 2019-03-07:
@gkasprow I'll have some time this evening to review schematics. Do you think they'll be ready by then.

Also, there are a few outstanding Sayma issues in this repo. Can they be closed now?

hartytp, 2019-03-11:
@jbqubit can we close this issue?

closed 2019-03-11 by hartytp