sinara-hw/sinara #606: Sayma power budget

gkasprow, 2018-12-19:
[power consumption.xlsx](https://github.com/sinara-hw/sinara/files/2696566/power.consumption.xlsx)

jbqubit, 2018-12-19:
Is there enough headroom for these supplies? 
- On RTM P6V0 there's 218 mA headroom (3%). 
- On AMC P1V0 there's 160 mA headroom (5%). 
- On AMC P1V5 there's 100 mA headroom (3%).
- On Metlino P1V0 there's 160 mA headroom (5%).

marmeladapk, 2018-12-19:
@jbqubit This is probably maximum current value, that could be found in data sheets, real consumption will be always lower than that. I'll check it tommorow.

marmeladapk, 2019-01-07:
@gkasprow This power budget for Sayma AMC and Metlino is outdated, it doesn't list new components and lists those that were removed.

hartytp, 2019-03-11:
@gkasprow I guess we won't want a power budget for Sayma on the schematic as we did with the EEMs, since it's too complicated. Will this go in the documentation or just in a spreadsheet in the repo? Either way, I guess it's a natural thing to do as part of the design review?

gkasprow, 2019-03-11:
Yes, this needs to be done for design review. We got rid of plenty of components so we won't exceed any current limits :)