sinara-hw/sinara #608: most NATview functions are broken with Sayma

sbourdeauducq, 2019-01-09:
e.g. when double-clicking on "Sayma" in the device list
[image]

Cannot guess if those are NAT bugs or Sayma MMC bugs...

hartytp, 2019-03-11:
I didn't have a very positive experience with NatView either when I played with it.

However, I'm not sure what the action point of this issue is. If we want to complain about the quality of NatView then we should talk to NAT. Ideally though, I guess we should aim for Sayma to work out of the box without a user having to touch NAT View.

sbourdeauducq, 2019-03-12:
I'm not sure if this is a NATview problem or a Sayma problem - those functions work correctly when applied to other modules, which suggests the latter.
NATview is useful to power-cycle Sayma remotely when it's in the crate.

marmeladapk, 2019-03-12:
@hartytp @sbourdeauducq As I said before in our lab (and in Duke) we didn't have any problems with NATview. @sbourdeauducq could send us an exact version that he used and we could try that in our lab. If it works correctly, then perhaps there are some patches that are not applied correctly or broke off?

@gkasprow are there any patches in v1 that change smth related to MMC or communication with MCH?