sinara-hw/sinara #609: intermittent JTAG bug in µTCA crate

sbourdeauducq, 2019-01-13:
```
Info : clock speed 5000 kHz
Info : TAP xc7.tap does not have IDCODE
Info : TAP xcu.tap does not have IDCODE
Info : TAP auto0.tap does not have IDCODE
Info : TAP auto1.tap does not have IDCODE
Info : TAP auto2.tap does not have IDCODE
Info : TAP auto3.tap does not have IDCODE
Info : TAP auto4.tap does not have IDCODE
Info : TAP auto5.tap does not have IDCODE
Info : TAP auto6.tap does not have IDCODE
Info : TAP auto7.tap does not have IDCODE
Info : TAP auto8.tap does not have IDCODE
Info : TAP auto9.tap does not have IDCODE
Info : TAP auto10.tap does not have IDCODE
Info : TAP auto11.tap does not have IDCODE
Info : TAP auto12.tap does not have IDCODE
Info : TAP auto13.tap does not have IDCODE
Info : TAP auto14.tap does not have IDCODE                                            
Info : TAP auto15.tap does not have IDCODE                                            
Info : TAP auto16.tap does not have IDCODE 
Info : TAP auto17.tap does not have IDCODE
Info : TAP auto18.tap does not have IDCODE                                        
Warn : Unexpected idcode after end of chain: 21 0x00000000                        
Warn : Unexpected idcode after end of chain: 53 0x00000000                    
Warn : Unexpected idcode after end of chain: 85 0x00000000                    
Warn : Unexpected idcode after end of chain: 117 0x00000000                  
Warn : Unexpected idcode after end of chain: 149 0x00000000                  
Warn : Unexpected idcode after end of chain: 181 0x00000000             
Warn : Unexpected idcode after end of chain: 213 0x00000000                 
Warn : Unexpected idcode after end of chain: 245 0x00000000                                      
Warn : Unexpected idcode after end of chain: 277 0x00000000                                      
Warn : Unexpected idcode after end of chain: 309 0x00000000                                                                                                                                                       Warn : Unexpected idcode after end of chain: 341 0x00000000                                                                                                                                                       Warn : Unexpected idcode after end of chain: 373 0x00000000                                                                                                                                                       Warn : Unexpected idcode after end of chain: 405 0x00000000  
Warn : Unexpected idcode after end of chain: 437 0x00000000                                                                                                                                                       
Warn : Unexpected idcode after end of chain: 469 0x00000000                                                                                                                                                       Warn : Unexpected idcode after end of chain: 501 0x00000000                                                                                                                                                       Warn : Unexpected idcode after end of chain: 533 0x00000000                                                                                                                                                       
Warn : Unexpected idcode after end of chain: 565 0x00000000
Warn : Unexpected idcode after end of chain: 597 0x00000000
Warn : Unexpected idcode after end of chain: 629 0x20000000
Error: double-check your JTAG setup (interface, speed, ...)
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : gdb server disabled

```

Only recoverable with a full power-cycle of the crate, rebooting the MCH or deactivating/reactivating Sayma do not help.

Reproduced on two different boards and never seen outside µTCA so I suspect this has to do with it.

gkasprow, 2019-01-13:
This is a known issue related with SCANSTA chip.
This is one of the reasons I decided to kick it away. For some reason, long uTCA backplane JTAG lines connected to SCANSTA chip break the communication.

hartytp, 2019-03-07:
Closing this issue since (1) it shouldn't be present in v2.0 since we're removing the SCANSTA chip (2) I don't think anyone is going to fix this for the v1.0 boards.

closed 2019-03-07 by hartytp